siemens/meta-iot2050 #225: How to integrate docker into the image

lxmuyu, 2021-12-06:
I installed the docker(18.09.1) and docker-compose（1.21.0） in the image(tag:V01.01.01), but cannot integrate them in .bb files.

.bb files:
docker-18.09.1.bb

```
DESCRIPTION = "docker"

inherit dpkg

SRC_URI="apt://docker.io=18.09.1"
```
docker-compose-1.21.0.bb
```
DESCRIPTION = "docker-compose"

inherit dpkg

SRC_URI="apt://docker-compose=1.21.0"

do_install_append() {
    install -v -d ${D}/etc/docker
    install -v -m 644 ${WORKDIR}/daemon.json ${D}/etc/docker/daemon.json
}
```

iot2050-image-example.bb
```

...
IMAGE_INSTALL += " \
    ...
    node-red-preinstalled-nodes \
    docker-18.09.1 \
    "

```
But I got the errors:
```
 Done
Starting pkgProblemResolver with broken count: 0
Starting 2 pkgProblemResolver with broken count: 0
Done
The following packages will be REMOVED:
  docker.io-cross-build-deps:arm64
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 10.2 kB disk space will be freed.
(Reading database ... 30384 files and directories currently installed.)
Removing docker.io-cross-build-deps:arm64 (20.10.5+dfsg1-1) ...
mk-build-deps: Unable to install docker.io-cross-build-deps at /usr/bin/mk-build-deps line 416.
mk-build-deps: Unable to install all build-dep packages
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/docker-18.09.1/1.0-r0/temp/log.do_install_builddeps.69
ERROR: Task (/repo/recipes-app/docker/docker-18.09.1.bb:do_install_builddeps) failed with exit code '1'
```

logfile:log.do_install_builddeps.69

[](url)
[log.do_install_builddeps.69.txt](https://github.com/siemens/meta-iot2050/files/7657793/log.do_install_builddeps.69.txt)

How can I fix it?

jan-kiszka, 2021-12-06:
How about `IMAGE_PREINSTALL += "docker.io"`? No need for a recipe, the binary Debian package works as is.

lxmuyu, 2021-12-06:
> How about `IMAGE_PREINSTALL += "docker.io"`? No need for a recipe, the binary Debian package works as is.

Yes,it works.Thanks!

closed 2021-12-06 by jan-kiszka