siemens/meta-iot2050 #233: xmit timeout issue

kanghan90, 2021-12-18:
We got an xmit timeout issue and ethernet stop communication, 
image built from V01.01.01
devie IOT2050 

Problem also mentioned in https://e2e.ti.com/support/processors-group/processors/f/processors-forum/955037/processor-sdk-am65x-ethernet-transmit-queue-timeouts

looks like recently fixed as TI forum mention

    [ I see this same failure signature mentioned in our internal bug-tracker in bug LCPD-19648 which I see is fixed in Linux SDK 7.01 which is due out shortly. That same bug report mentions it was seen specifically in SR2.0 silicon which seems to coincide with the reports here.  In short, I expect you will have a fix for this very shortly when SDK 7.1 is released.]

Can you please update stable version and make a release ?
We only use stable release and currently latest one is getting "xmit timeout issue" if Ethernet connection is used for 24 hours so

Thank you

AsuraZeng, 2021-12-20:
could you pls show your  board  information using below command
```
cat /etc/os-release
```

kanghan90, 2021-12-20:
root@iot2050-debian:~# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
BUILD_ID="V01.01.01-0-ge57a697"
VARIANT="IOT2050 Debian Example Image"
VARIANT_VERSION="1.0"

AsuraZeng, 2021-12-20:
OK
Do you have any specific error log messages?
how do you use it and cause this problem

AsuraZeng, 2021-12-20:
could you pls also show you board env using
```
fw_printenv
```

kanghan90, 2021-12-20:
root@iot2050-debian:~# fw_printenv
arch=arm
baudrate=115200
board=iot2050
board_a5e=A5E452229880AA05
board_name=IOT2050-ADVANCED
board_serial=N6E03284
board_uuid=8DAB1ACFEFE54B49AD512CBD9202FBB2
boot_a_script=load ${devtype} ${devnum}:${distro_bootpart} ${scriptaddr} ${prefix}${script}; source ${scriptaddr}
boot_efi_binary=load ${devtype} ${devnum}:${distro_bootpart} ${kernel_addr_r} efi/boot/bootaa64.efi; if fdt addr ${fdt_addr_r}; then bootefi ${kernel_addr_r} ${fdt_addr_r};else bootefi ${kernel_addr_r} ${fdtcontroladdr};fi
boot_efi_bootmgr=if fdt addr ${fdt_addr_r}; then bootefi bootmgr ${fdt_addr_r};else bootefi bootmgr;fi
boot_extlinux=sysboot ${devtype} ${devnum}:${distro_bootpart} any ${scriptaddr} ${prefix}${boot_syslinux_conf}
boot_net_usb_start=usb start
boot_pci_enum=pci enum
boot_prefixes=/ /boot/
boot_script_dhcp=boot.scr.uimg
boot_scripts=boot.scr.uimg boot.scr
boot_syslinux_conf=extlinux/extlinux.conf
boot_targets=mmc1 mmc0 usb0 usb1 usb2 pxe dhcp 
bootcmd=run distro_bootcmd
bootcmd_dhcp=setenv devtype dhcp; run boot_net_usb_start; run boot_pci_enum; run init_icssg0; if dhcp ${scriptaddr} ${boot_script_dhcp}; then source ${scriptaddr}; fi;setenv efi_fdtfile ${fdtfile}; setenv efi_old_vci ${bootp_vci};setenv efi_old_arch ${bootp_arch};setenv bootp_vci PXEClient:Arch:00011:UNDI:003000;setenv bootp_arch 0xb;if dhcp ${kernel_addr_r}; then tftpboot ${fdt_addr_r} dtb/${efi_fdtfile};if fdt addr ${fdt_addr_r}; then bootefi ${kernel_addr_r} ${fdt_addr_r}; else bootefi ${kernel_addr_r} ${fdtcontroladdr};fi;fi;setenv bootp_vci ${efi_old_vci};setenv bootp_arch ${efi_old_arch};setenv efi_fdtfile;setenv efi_old_arch;setenv efi_old_vci;
bootcmd_mmc0=devnum=0; run mmc_boot
bootcmd_mmc1=devnum=1; run mmc_boot
bootcmd_pxe=run boot_net_usb_start; run boot_pci_enum; run init_icssg0; dhcp; if pxe get; then pxe boot; fi
bootcmd_usb0=devnum=0; run usb_boot
bootcmd_usb1=devnum=1; run usb_boot
bootcmd_usb2=devnum=2; run usb_boot
bootdelay=2
cpu=armv8
distro_bootcmd=for target in ${boot_targets}; do run bootcmd_${target}; done
efi_dtb_prefixes=/ /dtb/ /dtb/current/
eth1addr=8c:f3:19:29:a7:f2
eth2addr=8c:f3:19:29:a7:f3
ethaddr=8c:f3:19:29:a7:f2
fdt_addr_r=0x82000000
fdtcontroladdr=fdf03100
fdtfile=ti/k3-am6548-iot2050-advanced.dtb
fw_version=2021.04-V01.01.01-0-ge57a697
init_icssg0=rproc init; sf probe; run load_icssg0_pru1_fw; run load_icssg0_rtu1_fw
kernel_addr_r=0x80080000
load_efi_dtb=load ${devtype} ${devnum}:${distro_bootpart} ${fdt_addr_r} ${prefix}${efi_fdtfile}
load_icssg0_pru1_fw=sf read $loadaddr 0x7d0000 0x8000; rproc load 4 $loadaddr 0x8000
load_icssg0_rtu1_fw=sf read $loadaddr 0x7f0000 0x8000; rproc load 5 $loadaddr 0x8000
loadaddr=0x80080000
mlfb=6ES7647-0BA00-1YA2
mmc_boot=if mmc dev ${devnum}; then devtype=mmc; run scan_dev_for_boot_part; fi
overlay_addr_r=0x83000000
pxefile_addr_r=0x80080000
ramdisk_addr_r=0x81000000
scan_dev_for_boot=echo Scanning ${devtype} ${devnum}:${distro_bootpart}...; for prefix in ${boot_prefixes}; do run scan_dev_for_extlinux; run scan_dev_for_scripts; done;run scan_dev_for_efi;
scan_dev_for_boot_part=part list ${devtype} ${devnum} -bootable devplist; env exists devplist || setenv devplist 1; for distro_bootpart in ${devplist}; do if fstype ${devtype} ${devnum}:${distro_bootpart} bootfstype; then run scan_dev_for_boot; fi; done; setenv devplist
scan_dev_for_efi=setenv efi_fdtfile ${fdtfile}; for prefix in ${efi_dtb_prefixes}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${efi_fdtfile}; then run load_efi_dtb; fi;done;run boot_efi_bootmgr;if test -e ${devtype} ${devnum}:${distro_bootpart} efi/boot/bootaa64.efi; then echo Found EFI removable media binary efi/boot/bootaa64.efi; run boot_efi_binary; echo EFI LOAD FAILED: continuing...; fi; setenv efi_fdtfile
scan_dev_for_extlinux=if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${boot_syslinux_conf}; then echo Found ${prefix}${boot_syslinux_conf}; run boot_extlinux; echo SCRIPT FAILED: continuing...; fi
scan_dev_for_scripts=for script in ${boot_scripts}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${script}; then echo Found U-Boot script ${prefix}${script}; run boot_a_script; echo SCRIPT FAILED: continuing...; fi; done
scriptaddr=0x83000000
seboot_version=V01.00.00.3-2-gb3f17ac6-0x0000
soc=k3
usb_boot=usb start; if usb dev ${devnum}; then devtype=usb; run scan_dev_for_boot_part; fi
usb_pgood_delay=900
vendor=siemens

kanghan90, 2021-12-20:
> OK Do you have any specific error log messages? how do you use it and cause this problem

icssg-prueth pruss0_eth eno2: xmit timout

AsuraZeng, 2021-12-20:
I find the uboot version your are used is:
```
fw_version=2021.04-V01.01.01-0-ge57a697
```

I wonder how do your upgrade your uboot to this version?

kanghan90, 2021-12-20:
I got the one of the latest code and upgraded so which version is the stable ?

AsuraZeng, 2021-12-20:
you build the artifacts by yourself not using the one in the siemens forum?
which tool do you use to upgrade the uboot?

kanghan90, 2021-12-20:
nope, I got the source code from this repo and used SD card way to upgrade

# Building and programming the boot loader

## Building the image

The boot loader is built like this:

```shell
./kas-container build kas-iot2050-boot.yml
```

After the build the boot image is under

```text
build/tmp/deploy/images/iot2050/iot2050-image-boot.bin
```

## Flashing the image

> :warning:
> Flashing an incorrect image may brick the device!

Write `iot2050-image-boot.bin` to an SD card and insert that into
the target device. Then boot into the U-Boot shell and execute there:

```shell
sf probe
load mmc 0:1 $loadaddr /path/to/iot2050-image-boot.bin
sf update $loadaddr 0x0 $filesize
```

> :note:
> When updating the boot loader of the BASIC variant, make sure to remove
> 0014-iot2050-Provide-dtb-for-devices-using-boot-load-V01..patch from the kernel
> patch queue in recipes-kernel/linux/linux-iot2050_*.bb.

## Recovering a bricked device

If the device does not come up anymore after flashing the boot loader, you can
recover it with the help of an external flash programmer. Known to work are the
Dediprog SF100 or SF600. Attach the programmer to X17, then run the following
on the host machine:

```shell
dpcmd --vcc 2 -v -u iot2050-image-boot.bin
```

kanghan90, 2021-12-20:
anyway we will fall back to 01.00.03 for time being

BaochengSu, 2021-12-21:
@kanghan90 

At the time being the current latest mater is muture enough to do the next release V01.02.01. If no surprise the next V release will be tagged within this week, or no later than next week due to the Christmas holidy...

The only remaining work is a few trivial things related to PG2 boards. I suppose you are holding a PG1 device, so I would build on the latest master branch if I were you.

IMHO, V01.00.03, which is almost the same as V01.00.02, is too old for new applications to use.

BaochengSu, 2021-12-21:
Back to this issue, iirc this is a known issue of the tripple one version, please correct me @AsuraZeng

jan-kiszka, 2023-10-23:
I suppose this can be closed by now, many updates since then.

closed 2023-10-23 by jan-kiszka