siemens/meta-iot2050 #238: modprobe: FATAL: Module br_netfilter not found in directory /lib/modules/5.10.64

kanghan90, 2022-01-05:
Fresh docker installation failure

Error =
WARNING: bridge-nf-call-iptables is disabled
WARNING: bridge-nf-call-ip6tables is disabled

tried to install the kernel driver but missing
modprobe br_netfilter
=
modprobe: FATAL: Module br_netfilter not found in directory /lib/modules/5.10.64

Reason =
missing kernel driver ! 

cat /etc/os-release
=
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
BUILD_ID="V01.01.01-111-gf9615cf"
VARIANT="IOT2050 Debian Example Image"
VARIANT_VERSION="1.0"
root@iot2050-debian:~# 
###################################################

fw_printenv
=
arch=arm
baudrate=115200
board=iot2050
board_a5e=A5E452229880AA05
board_name=IOT2050-ADVANCED
board_serial=N6E03284
board_uuid=8DAB1ACFEFE54B49AD512CBD9202FBB2
boot_a_script=load ${devtype} ${devnum}:${distro_bootpart} ${scriptaddr} ${prefix}${script}; source ${scriptaddr}
boot_efi_binary=load ${devtype} ${devnum}:${distro_bootpart} ${kernel_addr_r} efi/boot/bootaa64.efi; if fdt addr ${fdt_addr_r}; then bootefi ${kernel_addr_r} ${fdt_addr_r};else bootefi ${kernel_addr_r} ${fdtcontroladdr};fi
boot_efi_bootmgr=if fdt addr ${fdt_addr_r}; then bootefi bootmgr ${fdt_addr_r};else bootefi bootmgr;fi
boot_extlinux=sysboot ${devtype} ${devnum}:${distro_bootpart} any ${scriptaddr} ${prefix}${boot_syslinux_conf}
boot_fdt=try
boot_net_usb_start=usb start
boot_pci_enum=pci enum
boot_prefixes=/ /boot/
boot_script_dhcp=boot.scr.uimg
boot_scripts=boot.scr.uimg boot.scr
boot_syslinux_conf=extlinux/extlinux.conf
boot_targets=mmc1 mmc0 usb0 usb1 usb2 
bootcmd=run distro_bootcmd
bootcmd_mmc0=devnum=0; run mmc_boot
bootcmd_mmc1=devnum=1; run mmc_boot
bootcmd_usb0=devnum=0; run usb_boot
bootcmd_usb1=devnum=1; run usb_boot
bootcmd_usb2=devnum=2; run usb_boot
bootdelay=2
bootm_size=0x10000000
cpu=armv8
distro_bootcmd=for target in ${boot_targets}; do run bootcmd_${target}; done
dtboaddr=0x89000000
efi_dtb_prefixes=/ /dtb/ /dtb/current/
eth1addr=8c:f3:19:29:a7:f3
eth2addr=8c:f3:19:29:a7:f2
fdt_addr_r=0x88000000
fdtaddr=0x88000000
fdtcontroladdr=fdf06310
fdtfile=ti/k3-am6548-iot2050-advanced.dtb
fw_version=2021.04-V01.01.01-111-gf9615cf
kernel_addr_r=0x82000000
load_efi_dtb=load ${devtype} ${devnum}:${distro_bootpart} ${fdt_addr_r} ${prefix}${efi_fdtfile}
loadaddr=0x82000000
mlfb=6ES7647-0BA00-1YA2
mmc_boot=if mmc dev ${devnum}; then devtype=mmc; run scan_dev_for_boot_part; fi
pxefile_addr_r=0x80100000
ramdisk_addr_r=0x88080000
rdaddr=0x88080000
scan_dev_for_boot=echo Scanning ${devtype} ${devnum}:${distro_bootpart}...; for prefix in ${boot_prefixes}; do run scan_dev_for_extlinux; run scan_dev_for_scripts; done;run scan_dev_for_efi;
scan_dev_for_boot_part=part list ${devtype} ${devnum} -bootable devplist; env exists devplist || setenv devplist 1; for distro_bootpart in ${devplist}; do if fstype ${devtype} ${devnum}:${distro_bootpart} bootfstype; then run scan_dev_for_boot; fi; done; setenv devplist
scan_dev_for_efi=setenv efi_fdtfile ${fdtfile}; for prefix in ${efi_dtb_prefixes}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${efi_fdtfile}; then run load_efi_dtb; fi;done;run boot_efi_bootmgr;if test -e ${devtype} ${devnum}:${distro_bootpart} efi/boot/bootaa64.efi; then echo Found EFI removable media binary efi/boot/bootaa64.efi; run boot_efi_binary; echo EFI LOAD FAILED: continuing...; fi; setenv efi_fdtfile
scan_dev_for_extlinux=if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${boot_syslinux_conf}; then echo Found ${prefix}${boot_syslinux_conf}; run boot_extlinux; echo SCRIPT FAILED: continuing...; fi
scan_dev_for_scripts=for script in ${boot_scripts}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${script}; then echo Found U-Boot script ${prefix}${script}; run boot_a_script; echo SCRIPT FAILED: continuing...; fi; done
scriptaddr=0x80000000
seboot_version=V01.02.01.04-0-g398b965c-0x0000
soc=k3
usb_boot=usb start; if usb dev ${devnum}; then devtype=usb; run scan_dev_for_boot_part; fi
usb_pgood_delay=900
vendor=siemens

jan-kiszka, 2022-01-05:
Which docker version from which source are you installing. Is it only warning about a missing optional feature or is it actually failing?

But feel free to propose CONFIG_BRIDGE_NETFILTER as addition to our https://github.com/siemens/meta-iot2050/blob/master/recipes-kernel/linux/files/iot2050_defconfig_extra.cfg as pull request if that resolves the issue.

kanghan90, 2022-01-06:
tried all given solutions online still not working, same installation works in 01.01.01v stable release but 01.01.01v has xmit issue(already post the bug) also can not use please check it thank you

root@iot2050-debian:~# docker version
=
Client: Docker Engine - Community
 Version:           20.10.12
 API version:       1.41
 Go version:        go1.16.12
 Git commit:        e91ed57
 Built:             Mon Dec 13 11:44:32 2021
 OS/Arch:           linux/arm64
 Context:           default
 Experimental:      true

Server: Docker Engine - Community
 Engine:
  Version:          20.10.12
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.16.12
  Git commit:       459d0df
  Built:            Mon Dec 13 11:43:07 2021
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.4.12
  GitCommit:        7b11cfaabd73bb80907dd23182b9347b4245eb5d
 runc:
  Version:          1.0.2
  GitCommit:        v1.0.2-0-g52b36a2
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0

############################################

root@iot2050-debian:~# docker run hello-world
=
docker: Error response from daemon: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: process_linux.go:508: setting cgroup config for procHooks process caused: bpf_prog_query(BPF_CGROUP_DEVICE) failed: function not implemented: unknown.
ERRO[0000] error waiting for container: context canceled 

############################################

root@iot2050-debian:~# docker info
=
Client:
 Context:    default
 Debug Mode: false
 Plugins:
  app: Docker App (Docker Inc., v0.9.1-beta3)
  buildx: Docker Buildx (Docker Inc., v0.7.1-docker)
  compose: Docker Compose (Docker Inc., v2.2.2)

Server:
 Containers: 2
  Running: 0
  Paused: 0
  Stopped: 2
 Images: 2
 Server Version: 20.10.12
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 io.containerd.runtime.v1.linux runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 7b11cfaabd73bb80907dd23182b9347b4245eb5d
 runc version: v1.0.2-0-g52b36a2
 init version: de40ad0
 Security Options:
  seccomp
   Profile: default
  cgroupns
 Kernel Version: 5.10.64
 Operating System: Debian GNU/Linux 11 (bullseye)
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 1.889GiB
 Name: iot2050-debian
 ID: 4FDI:AVD3:ILNH:GI5Z:RBIG:XVTZ:Q2AQ:ILCO:CATT:LXFC:RSNL:ILA5
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

WARNING: bridge-nf-call-iptables is disabled
WARNING: bridge-nf-call-ip6tables is disabled

################################################
root@iot2050-debian:~# dockerd 
=
INFO[2022-01-06T03:03:19.383509999Z] Starting up                                  
failed to start daemon: pid file found, ensure docker is not running or delete /var/run/docker.pid

################################################

root@iot2050-debian:~# systemctl status docker
=
● docker.service - Docker Application Container Engine
     Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset>
     Active: active (running) since Wed 2022-01-05 02:55:12 UTC; 24h ago
TriggeredBy: ● docker.socket
       Docs: https://docs.docker.com
   Main PID: 388 (dockerd)
      Tasks: 11
     Memory: 113.2M
        CPU: 1min 22.693s
     CGroup: /system.slice/docker.service
             └─388 /usr/bin/dockerd -H fd:// --containerd=/run/containerd/conta>

Jan 05 02:55:12 iot2050-debian systemd[1]: Started Docker Application Container>
Jan 05 02:55:13 iot2050-debian dockerd[388]: time="2022-01-05T02:55:13.06693654>
Jan 05 02:56:34 iot2050-debian dockerd[388]: time="2022-01-05T02:56:34.71684125>
Jan 05 02:56:34 iot2050-debian dockerd[388]: time="2022-01-05T02:56:34.71724037>
Jan 05 02:56:34 iot2050-debian dockerd[388]: time="2022-01-05T02:56:34.99407039>
Jan 05 02:56:34 iot2050-debian dockerd[388]: time="2022-01-05T02:56:34.99443449>
Jan 06 03:02:00 iot2050-debian dockerd[388]: time="2022-01-06T03:02:00.12672354>
Jan 06 03:02:00 iot2050-debian dockerd[388]: time="2022-01-06T03:02:00.12691989>
Jan 06 03:02:00 iot2050-debian dockerd[388]: time="2022-01-06T03:02:00.29630011>
Jan 06 03:02:00 iot2050-debian dockerd[388]: time="2022-01-06T03:02:00.29657595>
lines 1-22/22 (END)

BaochengSu, 2022-01-06:
This is due to the CGROUP_BPF is not enabled, check [my branch](https://github.com/siemens/meta-iot2050/tree/su/cgroup-bpf), it fixes this issue.

BaochengSu, 2022-01-06:
@jan-kiszka  

Seems the `docker.io` package maintained by debian is still at verison 20.10.6, and when running it a warning message shows that the CGROUPV2 support is experimental.

Meanwhile the docker community version is 20.10.12, and the CGROUPV2 warning has been gone, seems it's matured in this version.

BaochengSu, 2022-01-06:
@kanghan90 

To save time, you can just build the kernel package by:

```
./kas-container shell kas-iot2050-example.yml
```

Then in the container shell:

```
bitbake linux-iot2050
```
Then copy the `linux-image-iot2050_5.10.64+r0_arm64.deb` under `build/tmp/work/iot2050-debian-arm64/linux-iot2050/5.10.64-r0` to the IOT2050 device and use 'dpkg -i linux-image-iot2050_5.10.64+r0_arm64.deb && reboot' to install it and reboot to take effect.

kanghan90, 2022-01-06:
Company policy only stable release can be given to customer so when v01.01.04 will be released ?

BaochengSu, 2022-01-06:
The upcoming stable release is V01.02.01 and has been freezed internally on the latest master, so after the testing team finished their job, the V01.02.01 release will be tagged on the latest master, If everything goes smoothly then I think tomorrorw will be the expected day to tag that.

Meanwhile, after PG2 release, the expected release cadence is 6 month. Back to this issue, basically it's too late for it goes into V01.02.01, so I think the feasible plan is to integrate it into the next release half year later.

Then back to the term `stable`, the current master branch is considered as stable enough given that the testing team has been test against it iterately multiple times. And AFAIK, the docker functionality is out of the scope of the test plan, due to the docker is not integrated into the example image by default, so if the plan is to integrate this patch, then additional work need to be performed internally to add test cases to cover this use case.

If half year is too long for you to wait, could you consider to perform some integrating test on your side? I assume your application is a concrete one and need not provide an open platform to your costomer? If so I think the test scope should be limited on your side and it would not take too much time to finish it.

jan-kiszka, 2022-01-06:
I think we should add
```
apt install docker.io
docker run debian:bullseye-slim
```
to our test plan although we don't ship that package(s) pre-installed. This is too important.

kanghan90, 2022-01-06:
Device is not open platform to customer but getting installed into big factories with safe zone requirements so in case of OS issue, fixing the device is getting very complicated but for this time will ask to use stable + fix.

jan-kiszka, 2022-01-06:
If you have security requirements to fulfill, you need to maintain the image yourself anyway - half a year gap between two releases is generally too long for security hot-fixes. Therefore, this here is a reference image, not a product. And as you add docker yourself already, you are customizing significantly, thus have to do own extensive integration tests.

BaochengSu, 2022-01-06:
> I think we should add
> 
> ```
> apt install docker.io
> docker run debian:bullseye-slim
> ```
> 
> to our test plan although we don't ship that package(s) pre-installed. This is too important.

Fully agree.

jan-kiszka, 2022-01-06:
...and probably also another `apt update` inside that container, to validate connectivity.

BaochengSu, 2022-01-06:
@kanghan90 

After some internal discussion we plan to do a hotfix release (expected as V01.02.02) to cover this fix, and we will do the test to guarantee that the `debian:bullseye-slim` works as expected as Jan suggested.

I assume this hotfix version would not take too long, maybe within month condering the project management overhead.

Meanwhile please consider the suggestion from Jan regarding how to use this example image.

BaochengSu, 2022-01-06:
> ...and probably also another `apt update` inside that container, to validate connectivity.

`apt install docker.io` this installs the 20.10.5 which may not work w/ CGROUP2... Or do you know any news regaring this topic?

jan-kiszka, 2022-01-06:
Our kernel should enable everything needed to run the official Debian docker package.

BaochengSu, 2022-01-06:
@kanghan90 

...BTW we will test only on the [`docker.io`](https://packages.debian.org/bullseye/docker.io) package, this is the one maintained by debian directly.

jan-kiszka, 2022-02-28:
Fixed in master.

closed 2022-02-28 by jan-kiszka