siemens/meta-iot2050 #246: Support for NodeJS > 12

Webbeh, 2022-01-13:
Currently, the installed NodeJS version is 12.x. Support for that version is due to end in mid-april 2022, and it would be good to upgrade to something more up-to-date.

Everything is pretty much upgradable by adding the correct debian repository (using the handy https://deb.nodesource.com/setup_17.x scripts provided by NodeJS) and updating node via apt.

Everything works fine on the Node side, as well as node-red, with the exception of MRAA. Its node module requires a recompilation against the newer NodeJS version, but currently fails miserably.

I also tried to compile MRAA from the [eclipse/mraa](https://github.com/eclipse/mraa/) source. Although the compilation seems to finish properly, it doesn't seem to provide me with a node module. I'm probably doing something wrong but didn't have the courage to dig further into it so far.

I don't know if going the MRAA route is the best option. Their latest NPM package is from 4 years ago, and since its switch from intel to eclipse, I don't see it getting updated anytime soon. All that being said, I have no clue what to take instead.

jan-kiszka, 2022-01-13:
Just one note regarding LTS support: Update NodeJS may stop maintenance of 12 this year, distros like Debian who are shipping that version will surely not. So the only reason to upgrade to newer major releases would be the need for newer features.

jan-kiszka, 2023-06-19:
We have Debian 12 support under testing, see #456. That will bring NodeJS 18.

jan-kiszka, 2023-07-19:
Merged into master now.

closed 2023-07-19 by jan-kiszka