siemens/meta-iot2050 #248: Wireguard installation

edu-sistec, 2022-01-17:
[image]
Hi,

I'm running this image with good results, but for me it's very important to be able to work Wireguard on this image and device, I'm having some compatibility issues.
I tried to install Wireguard using '_apt wireguard_' with errors. I tried to install Wireguard manually:

_apt install wireguard wireguard-tools linux-headers - $ (uname -r)_
No results, problems with the header version, image attached

I tried to do the automatic installation dedicated to Debina 11:
_curl -O https://raw.githubusercontent.com/angristan/wireguard-install/master/wireguard-install.sh_

and

_chmod + x wireguard-install.sh
./wireguard-install.sh_

In this last option it can be installed but at the time of starting the program it tells me that it cannot start the wireguard interface, in the attached screenshot you can see that I am trying to start the program. Looks like there's a problem setting up the Kernel? Does the Kernel seem to not allow access to a wireguard?

I can't find a solution to this problem. Do you know how to make Kernel and Wireguard work?

Thanks,
Edu

jan-kiszka, 2022-01-17:
Already `CONFIG_WIREGUARD` is currently not set in our kernel. Maybe there is more needed. You can find out if this along would be enough by enabling it in `recipes-kernel/linux/files/iot2050_defconfig_extra.cfg` and building your own kernel (does not have to be a complete image, `kas-container build kas-iot2050-example.yml --target linux-iot2050` is enough, find the .deb then in `build/tmp/work/iot2050-debian-arm64/linux-iot2050/5.10.64-r0/`). Bonus for submitting a PR with the needed configs afterwards.

edu-sistec, 2022-01-17:
Hi Jan,

Gràcies per la informació. Jo he fet el canvi i ha funcionat perfectement. ja tinc el Wireguard arrencat. 
Adjunto captura pantalla.

Ara intentare provar tots els recursos de Wireguard per saber si tot funciona. Si jo veig alguna cosa t'aviso, pero sembla que no haig de tindre problemes.

Best Reguards,
Edu
[image]

jan-kiszka, 2022-01-17:
Sorry I'm not speaking... Catalan? Others here as well. Google says you were successful, but could you still re-write in English?

commit c2cb6472 (2022-01-26): linux-iot2050: Enable Wireguard support

This has been requested as feature by users, so enable it as module.

Closes #248
commit 6cdf837a (2022-01-26): linux-iot2050: Enable Wireguard support

This has been requested as feature by users, so enable it as module.

Closes #248
closed 2022-01-26 by BaochengSu