siemens/meta-iot2050 #250: Digital input node from Siemens cusing Node-Red to abort

MohammadKurdia, 2022-01-18:
I have a configuration where 6 digital inputs are connected to 6 digital outputs. 

When I turn on and off the digital output, I read the signal in node red using the Digital input node provided from Siemens. 

But after some minutes of running this flow, Node-Red aborts for some reason !!

[image]

I replaced these digital input nodes with exec nodes with "Mraa" commands in them. and now it's working fine.

MohammadKurdia, 2022-01-24:
Example image V1.1.1

jan-kiszka, 2022-01-24:
Can you re-check with V1.2.1 as well? Note that this requires a firmware update as well.

MohammadKurdia, 2022-01-27:
Where can I find the V1.2.1 ?

jan-kiszka, 2022-01-27:
Not yet uploaded apparently... Please request via forum, direct message to bergmanu. Both the image and the firmware update package.

jan-kiszka, 2023-07-19:
I suppose this was resolved by now.

closed 2023-07-19 by jan-kiszka