siemens/meta-iot2050 #251: very slow boot

federico-d, 2022-01-22:
On v01.01.01 the example image boots very slowly due to crng init:

[   10.066721] pru-rproc b206000.rtu: PRU rproc node rtu@b206000 probed successf                                                                                                                                                             ully
[   10.279217] icssg-prueth pruss0_eth: TI PRU ethernet driver initialized: dual                                                                                                                                                              EMAC mode
[   33.846080] tps62363-vout: disabling
[  136.918242] random: crng init done
[  136.921663] random: 7 urandom warning(s) missed due to ratelimiting

is there a way to speed it up?
Thanks

jan-kiszka, 2022-01-24:
Is this without any network attached? Does the issue vanish when bringing the device online?

Was this a self-built image? Same issue when using the release binary version? Also, can you reproduce this with the latest image release (make sure to update the firmware for that first)?

federico-d, 2022-01-24:
> Is this without any network attached? Does the issue vanish when bringing the device online?
It doesn't matter, it happens at every boot.
> 
> Was this a self-built image? Same issue when using the release binary version? Also, can you reproduce this with the latest image release (make sure to update the firmware for that first)?
Yes, it is a self built image. I solved it (temporarily?) by adding -haveged- debian package that seems to speed up entropy accumulation.
One thing I noticed, the ti rng doesn't show up in the device tree and looks disabled.

jan-kiszka, 2022-01-24:
Is this an advanced or a basic variant?

And, again, do you get the same behavior when using the binary images that was released? I'm still confused that he haven't seen that before.

federico-d, 2022-01-24:
Its a basic version. I never used the stock image, could you point me to
the download link?

jan-kiszka, 2022-01-24:
Sure: https://support.industry.siemens.com/cs/document/109741799/downloads-for-simatic-iot20x0?dti=0&lc=en-WW#!

jan-kiszka, 2023-06-19:
Probably solved by now.

closed 2023-06-19 by jan-kiszka