siemens/meta-iot2050 #262: Error integrating docker （branch:remotes/origin/stable/V01.02）

lxmuyu, 2022-02-10:
When I  integrated docker.io into source, using 
```
IMAGE_PREINSTALL += " \
...
docker.io \
    "
```
, I got the error:
```

Configuration file '/etc/hosts'
 ==> File on system created by you or by a script.
 ==> File also in package provided by package maintainer.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** hosts (Y/I/N/O/D/Z) [default=N] ? dpkg: error processing package customizations-base (--configure):
 end of file on stdin at conffile prompt

...

done.
done.
Processing triggers for dbus (1.12.20-2) ...
Processing triggers for initramfs-tools (0.140) ...
update-initramfs: Generating /boot/initrd.img-5.10.64
Errors were encountered while processing:
 customizations-base
E: Sub-process /usr/bin/dpkg returned an error code (1)
WARNING: exit code 100 from a shell command.

DEBUG: Python function do_rootfs_install finished
[log.do_rootfs_install.97.txt](https://github.com/siemens/meta-iot2050/files/8039172/log.do_rootfs_install.97.txt)

```

How can I fix it ?

lxmuyu, 2022-02-10:
[image]

When I modifie `customizations-base_0.1-iot2050-debian.bb` file , delete :
```
install -v -m 644 ${WORKDIR}/hosts ${D}/etc/
```

compile is ok....
[image]

jan-kiszka, 2022-02-10:
We likely need to move /etc/hosts out of the ownership of customizations-base while still filling it (i.e. move it to the postinst script). Things worked so far when installing docker manually on the device. Your case was apparently not tested so far, but it is valid. Interestingly, docker.io will not come to a different /etc/hosts than customizations-base.

lxmuyu, 2022-02-10:
Docker.io should not affect /etc/hosts, but the error came.
Beside Docker.io, I has installed mariadb-server,nginx,logrotate and docker-compose. Only the Docker.io made error, It's strange.

lxmuyu, 2022-02-10:
Docker.io should not affect /etc/hosts, but the error came.
Beside Docker.io, I has installed mariadb-server,nginx,logrotate and docker-compose. Only the Docker.io made error, It's strange.

The same softs were added to the tag V01.01.01.That is all fine....

lxmuyu, 2022-02-10:
> We likely need to move /etc/hosts out of the ownership of customizations-base while still filling it (i.e. move it to the postinst script). Things worked so far when installing docker manually on [...]

Docker.io should not affect /etc/hosts, but the error came.
Beside Docker.io, I has installed mariadb-server,nginx,logrotate and docker-compose. Only the Docker.io made error, It's strange.

The same softs were added to the tag V01.01.01.That is all fine....

jan-kiszka, 2022-02-10:
Let me look into that... Should be quick to resolve, and the I will propose that for the stable branch as well.

PR #263 (2022-02-11): customizations-base: Do not package /etc/hosts, modify it
This avoids conflicts with other packages that want to adjust this file,
e.g. docker.io. And it's simpler because we can now let netbase deliver
the foundation.

This should resolve #262 (once merged into stable as well).
closed 2022-02-11 by BaochengSu