siemens/meta-iot2050 #272: Bug in Building and Image

gxcora, 2022-02-28:
Hi:

I get this error when building an LXDE image:
```
Summary: 4 tasks failed:
  /repo/recipes-app/node-red-contrib-modbus/node-red-contrib-modbus_5.14.1.bb:do_dpkg_build
  /repo/recipes-app/node-red-dashboard/node-red-dashboard_2.30.0.bb:do_dpkg_build
  /repo/recipes-app/mindconnect-node-red-contrib-mindconnect/mindconnect-node-red-contrib-mindconnect_3.12.0.bb:do_dpkg_build
  /repo/recipes-app/node-red-contrib-s7/node-red-contrib-s7_3.0.0.bb:do_dpkg_build
Summary: There were 4 ERROR messages shown, returning a non-zero exit code.
2022-02-28 06:43:31 - ERROR    - Command returned non-zero exit status 1
```
I've attached the full error, if you're interested. Have you seen this error before? Thanks

[errors.txt](https://github.com/siemens/meta-iot2050/files/8152399/errors.txt)

jan-kiszka, 2022-02-28:
Likely same or at least similar issue like in https://github.com/siemens/meta-iot2050/issues/219.

gxcora, 2022-02-28:
cool. let me know if you want me to try anything

jan-kiszka, 2022-02-28:
Please compare your build setup to the one mentioned there, specifically check if your have qemu-user-static installed and remove it if so (and reboot after that).

gxcora, 2022-02-28:
thanks Jan. that worked

closed 2022-02-28 by gxcora