siemens/meta-iot2050 #276: IOT2050 Advanced gets stuck during boot after firmware update

bergmanu, 2022-03-09:
When updating the firmware to V1.2.1 on IOT2050 Advanced from SD card with Example Image V1.1.1 while having **Example Image V1.0.2 or Industrial OS V2.x installed on the eMMc**, it is not possible to boot the IOT2050 Advanced after the firmware update. 

Reason: The boot order is reset by the firmware update, so eMMC is first boot device. When eMMc contains the **Example Image V1.0.2** the boot gets stuck (Logfile attached)
[iot2050_bootlog.txt](https://github.com/siemens/meta-iot2050/files/8214144/iot2050_bootlog.txt)

jan-kiszka, 2022-03-10:
At least, there should be a big warning and extra hints in this case: Users can recover from that by holding the USER button down during reboot, thus skipping the eMMC.
@BaochengSu

bergmanu, 2022-03-10:
Good idea, didn't think of the USER button to skip eMMc

jan-kiszka, 2023-07-19:
I think the button resolves this.

closed 2023-07-19 by jan-kiszka