siemens/meta-iot2050 #277: Using SPI Flash Programmer

delarosa1962, 2022-03-09:
I have to upgrade many IOT 2050 to F/W version 1.2.1 -- the current process is rather lengthy and would like to have all the steps to use the SPI Programmer.  What Image do I use ? What is the exact process ?

delarosa1962, 2022-03-09:
> Did you study https://github.com/siemens/meta-iot2050/blob/master/recipes-bsp/u-boot/README.md already? It describes the steps needed for programming externally (if that is what you mean with "use the SPI Programmer").

Yes, that is what I was looking for. X17 header. Then the question is the file format ? There are two types of images that could be sent, the firmware itself and the OS. Could both be done using the DediProg programmer ?

closed 2022-03-09 by delarosa1962

jan-kiszka, 2022-03-10:
X17 only attaches to the QSPI flash.

delarosa1962, 2022-03-28:
Hi, I purchased the SF 100, the X17 has 8 pins but the SF 100 cable has 14.  What specific cable part # do I need to get to be able to flash the f/w ?

lifang0504, 2022-03-29:
hello i think you need one of this cable , can transfer 14 pins to 8 pins. 

[image]

delarosa1962, 2022-03-29:
Yes, thanks. I ordered one, part number is:
https://www.dediprog.com/product/ISP1-CB