siemens/meta-iot2050 #280: Iot2050 Firmware update via swupdate fails

amitsagargowda, 2022-03-11:
We are trying swupdate on our iot2050 using the example mentioned in iot meta-2050 readme.md "$ swupdate -i iot2050-image-swu-example-iot2050-debian-iot2050.swu" .

But this fails with an error "Error: Not Enough free Space to extract img1.gz " . 

We are trying on PG1 FS03 using 32GB SD card. Base os is updated to commit id "78a1eaf924579aca87e26d6f511ecc0b660f0515".

We have updated event Bootloader to latest one compitable to PG2 OS. 

[image]

jan-kiszka, 2022-03-11:
@gylstorffq Where does swupdate extract the swu here, on /tmp or on the medium where it is picked up from?
@amitsagargowda What does `df -h` report in your case? Does it work when you first move the swu to /data or even /home/root and use it from there?

gylstorffq, 2022-03-11:
The swu binary is extracted to `/tmp`.

amitsagargowda, 2022-03-11:
@gylstorffq: is there a way to extract to different folder as there is not sufficient space is /tmp and currently it has only 312MB free.

I just followed the steps mentioned in README for swupdate

jan-kiszka, 2022-03-11:
That increase may relate to the addition of Google Coral support. You can disable that if you don't need it. Or increase the root partition sizes.

Otherwise, you can set `TMPDIR` before calling `swupdate` to make it use a different temp folder.

amitsagargowda, 2022-03-11:
@jan-kiszka: i tried setting TMPDIR and getting below error.

The /mnt/tmp is folder created in usb which has free space of 27GB.

[image]

[image]

jan-kiszka, 2022-03-11:
What's the filesystem of /dev/sda1? Some FAT?

gylstorffq, 2022-03-11:
I have build 78a1eaf924579aca87e26d6f511ecc0b660f0515. The image has around 1.5G free disk space on root. The commit doesn't contain meta-coral.

@amitsagargowda If you have installed additional software i would propose to adjust the disk sizes at  https://github.com/siemens/meta-iot2050/blob/78a1eaf924579aca87e26d6f511ecc0b660f0515/wic/iot2050-swu.wks.in#L14 and https://github.com/siemens/meta-iot2050/blob/78a1eaf924579aca87e26d6f511ecc0b660f0515/wic/iot2050-swu.wks.in#L17.

amitsagargowda, 2022-03-14:
@jan-kiszka : Yes the filesytem is of type FAT

amitsagargowda, 2022-03-15:
@jan-kiszka : Thanks for the update i did try formatting my usb to ext4 and not swupdate works logs can be seen in figure below. But after reboot it is not moving form one rootfs to other rootfs I am getting following error what is going wrong here.

Finally we see that it stills in same rootfs and not moved to flashed rootfs. Any support here will help us a lot.

We know that at the end we need to run script " complete_update.sh success" after we observe that rootfs has shifted correctly but since rootfs is not shifted we did not run script complete_update.sh. 
 

[image]

root@iot2050-debian:/media/usb0# TMPDIR=/media/usb0/tmp/ swupdate -v -i ./iot2050-image-swu-example-iot2050-debian-iot2050.swu
Swupdate v2021.04.0

Licensed under GPLv2. See source distribution for detailed copyright notices.

[INFO ] : SWUPDATE running :  [lua_handlers_init] : Lua handler(s) found.
[INFO ] : SWUPDATE running :  [print_registered_handlers] : Registered handlers:
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     dummy
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     archive
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     tar
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     uboot
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     bootloader
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     diskpart
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     uniqueuuid
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     lua
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     raw
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     rawfile
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     rawcopy
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     rdiff_image
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     rdiff_file
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     shellscript
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     preinstall
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     postinstall
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     roundrobin
[TRACE] : SWUPDATE running :  [listener_create] : got no socket at /media/usb0/tmp//swupdateprog from systemd
[TRACE] : SWUPDATE running :  [listener_create] : creating socket at /media/usb0/tmp//swupdateprog
[TRACE] : SWUPDATE running :  [network_initializer] : Main loop daemon
[TRACE] : SWUPDATE running :  [listener_create] : got no socket at /media/usb0/tmp//sockinstctrl from systemd
[TRACE] : SWUPDATE running :  [listener_create] : creating socket at /media/usb0/tmp//sockinstctrl
[TRACE] : SWUPDATE running :  [network_thread] : Incoming network request: processing...
[INFO ] : SWUPDATE started :  Software Update started !
[TRACE] : SWUPDATE running :  [network_initializer] : Software update started
[TRACE] : SWUPDATE running :  [extract_file_to_tmp] : Found file
[TRACE] : SWUPDATE running :  [extract_file_to_tmp] :   filename sw-description
[TRACE] : SWUPDATE running :  [extract_file_to_tmp] :   size 577
[TRACE] : SWUPDATE running :  [get_common_fields] : Version 0.1
[TRACE] : SWUPDATE running :  [add_properties] : Found properties for img1.gz:
[TRACE] : SWUPDATE running :  [add_properties_cb] :             Property subtype: image
[TRACE] : SWUPDATE running :  [_parse_images] : Found compressed Image: img1.gz in device : rootfs1,rootfs2 for handler roundrobin
[TRACE] : SWUPDATE running :  [extract_files] : Found file
[TRACE] : SWUPDATE running :  [extract_files] :         filename img1.gz
[TRACE] : SWUPDATE running :  [extract_files] :         size 1863127600 required
[TRACE] : SWUPDATE running :  [extract_padding] : Expecting 340 padding bytes at end-of-file
[TRACE] : SWUPDATE running :  [network_initializer] : Valid image found: copying to FLASH
[INFO ] : SWUPDATE running :  Installation in progress
[TRACE] : SWUPDATE running :  [install_single_image] : Found installer for stream img1.gz roundrobin
[TRACE] : SWUPDATE running :  [notify_helper] : Round Robin Handler: No compiled-in config found, trying /etc/swupdate.handler.ini
[INFO ] : SWUPDATE running :  [notify_helper] : Round Robin Handler: Using '/dev/disk/by-partlabel/rootfs2' as target via 'raw' handler.
[TRACE] : SWUPDATE running :  [lua_dump_table] : properties subtype = image
[TRACE] : SWUPDATE running :  [lua_dump_table] : Inserting property subtype = image
[INFO ] : SWUPDATE running :  [notify_helper] : Round Robin Handler: Setting bootloader environment: ustate=1
[TRACE] : SWUPDATE running :  [l_handler_wrapper] : [Lua handler] returned: 0
[INFO ] : SWUPDATE successful ! SWUPDATE successful !
[TRACE] : SWUPDATE running :  [network_initializer] : Main thread sleep again !
[INFO ] : No SWUPDATE running :  Waiting for requests...
[INFO ] : SWUPDATE running :  [endupdate] : Swupdate was successful !

jan-kiszka, 2022-03-15:
The watchdog error indicates that you are not using latest master for your tests. Correct? We fixed some issues there, and things worked after that, at least for me.

amitsagargowda, 2022-03-15:
@jan-kiszka : Which commit id should we refer. Currently we are using commit id [78a1eaf924579aca87e26d6f511ecc0b660f0515](https://github.com/siemens/meta-iot2050/commits/78a1eaf924579aca87e26d6f511ecc0b660f0515)

jan-kiszka, 2022-03-15:
master, bf54a81e1aa6bab868904cc4b66847e2e38c0ea4.

amitsagargowda, 2022-03-15:
@jan-kiszka : Currently we are using bootloader version mentioned below is this fine.

[image]

jan-kiszka, 2022-03-15:
It should be fine, yes.

There are many variants and versions by now, so we may always miss some corner case, specifically with rather fresh features. Please again with a vanilla image built from today's master. If that one shows the same error as before, I need to reproduce (don't have the exact same setup at hands right now).

amitsagargowda, 2022-03-16:
@jan-kiszka: I am trying to build image from commit id mentioned but i get following error. The build option used was " ./kas-container build kas-iot2050-example.yml" . Are we doing anything wrong here. 

[image]

jan-kiszka, 2022-03-16:
Please retry after running `./kas-container --isar clean` first.

amitsagargowda, 2022-03-17:
@jan-kiszka : I tried "./kas-container --isar clean" as suggested above and tried build options "./kas-container build kas-iot2050-example.yml" . But still i do get same error.

jan-kiszka, 2022-03-17:
Then please share the log file, rather than screenshots.

amitsagargowda, 2022-03-17:
@jan-kiszka : what log file would you expect . can you please let me know so i can upload it.

amitsagargowda, 2022-03-17:
Please find the logs files attached i hope this is what you are expecting.
[log.do_rootfs_install.279.txt](https://github.com/siemens/meta-iot2050/files/8282067/log.do_rootfs_install.279.txt)

jan-kiszka, 2022-03-17:
Double-check that you are not having any local modifications on top. E.g., I don't see why openssl-dev should be installed on the original version of bf54a81e1aa6bab868904cc4b66847e2e38c0ea4 (it's not installed when I build it here).

You might also be bitten by the new sstate cache and a Debian update of openssl in the meantime. That can be cleaned via `./kas-container --isar cleansstate`. If all that still does not help, I would need the console log (`./kas-container build kas-iot2050-example.yml > build.log`).

Note that you will want `./kas-container build kas-iot2050-swupdate.yml` for the SWUpdate-enabled image in the end.

amitsagargowda, 2022-03-17:
@jan-kiszka : I tried the cleansstate command but it seem to fail.  Are we entering correct command. 

[image]

jan-kiszka, 2022-03-17:
Nope, but the kas-container version in master is not recent enough. Just to `rm -r build/sstate-cache/` instead.

jan-kiszka, 2022-03-17:
We might indeed have a problem in the openssl recipe, just received another report. Hang on!

BaochengSu, 2022-03-17:
Our CI is broken now also due to the openssl issue.

BaochengSu, 2022-03-17:
@AsuraZeng already trigger a new CI to fix it, although that will not be the final solution.

jan-kiszka, 2022-03-17:
This will resolve the issue (for clean rebuilds):
```
diff --git a/recipes-security/openssl/openssl_latest.bb b/recipes-security/openssl/openssl_latest.bb
index d498c46e..f55e69b1 100644
--- a/recipes-security/openssl/openssl_latest.bb
+++ b/recipes-security/openssl/openssl_latest.bb
@@ -11,7 +11,7 @@
 inherit dpkg
 
 SRC_URI = " \
-    apt://${PN}/${BASE_DISTRO_CODENAME} \
+    apt://${PN} \
     file://0001-make-bnrand_range-reliable-with-deterministic-run-ti.patch;apply=no \
     "
 CHANGELOG_V="<orig-version>+iot2050"
```

jan-kiszka, 2022-03-17:
Specifically, a clean of the download cache for openssl seems to be needed (`./kas-container build kas-iot2050-example.yml --target openssl -c cleanall`, then rebuild).

amitsagargowda, 2022-03-17:
@jan-kiszka : Thanks a lot it works now 👍

amitsagargowda, 2022-03-22:
@jan-kiszka : Can you please confirm if i need to update the base image or u-boot for this swupdate feature to work?

jan-kiszka, 2022-03-22:
You need firmware 1.2.1, and I would strongly recommend to build from master.

Note that the SWUpdate feature should be fine in principle already, but some details will still need work. And the ongoing effort to enable secure boot for it may even bring more fundamental architectural changes (we are exploring if switching to UEFI wouldn't be better).

amitsagargowda, 2022-03-23:
@jan-kiszka : I build the firmware from master and updated it still i find the issue that rootfs is shifted still boots with old kernel i have attached detailed logs can you please have a look and mention what is going wrong here.

ebian-iot2050.swuan:/media/usb0# TMPDIR=/media/usb0/tmp/ swupdate -v -i ./iot2050-image-swu-example-iot2050-debian-iot2050.swu
Swupdate v2021.04.0

Licensed under GPLv2. See source distribution for detailed copyright notices.

[INFO ] : SWUPDATE running :  [lua_handlers_init] : Lua handler(s) found.
[INFO ] : SWUPDATE running :  [print_registered_handlers] : Registered handlers:
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     dummy
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     archive
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     tar
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     uboot
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     bootloader
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     diskpart
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     uniqueuuid
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     lua
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     raw
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     rawfile
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     rawcopy
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     rdiff_image
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     rdiff_file
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     shellscript
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     preinstall
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     postinstall
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     roundrobin
[TRACE] : SWUPDATE running :  [listener_create] : got no socket at /media/usb0/tmp//swupdateprog from systemd
[TRACE] : SWUPDATE running :  [listener_create] : creating socket at /media/usb0/tmp//swupdateprog
[TRACE] : SWUPDATE running :  [network_initializer] : Main loop daemon
[TRACE] : SWUPDATE running :  [listener_create] : got no socket at /media/usb0/tmp//sockinstctrl from systemd
[TRACE] : SWUPDATE running :  [listener_create] : creating socket at /media/usb0/tmp//sockinstctrl
[TRACE] : SWUPDATE running :  [network_thread] : Incoming network request: processing...
[INFO ] : SWUPDATE started :  Software Update started !
[TRACE] : SWUPDATE running :  [network_initializer] : Software update started
[TRACE] : SWUPDATE running :  [extract_file_to_tmp] : Found file
[TRACE] : SWUPDATE running :  [extract_file_to_tmp] :   filename sw-description
[TRACE] : SWUPDATE running :  [extract_file_to_tmp] :   size 577
[TRACE] : SWUPDATE running :  [get_common_fields] : Version 0.1
[TRACE] : SWUPDATE running :  [add_properties] : Found properties for img1.gz:
[TRACE] : SWUPDATE running :  [add_properties_cb] :             Property subtype: image
[TRACE] : SWUPDATE running :  [_parse_images] : Found compressed Image: img1.gz in device : rootfs1,rootfs2 for handler roundrobin
[TRACE] : SWUPDATE running :  [extract_files] : Found file
[TRACE] : SWUPDATE running :  [extract_files] :         filename img1.gz
[TRACE] : SWUPDATE running :  [extract_files] :         size 1863127600 required
[TRACE] : SWUPDATE running :  [extract_padding] : Expecting 340 padding bytes at end-of-file
[TRACE] : SWUPDATE running :  [network_initializer] : Valid image found: copying to FLASH
[INFO ] : SWUPDATE running :  Installation in progress
[TRACE] : SWUPDATE running :  [install_single_image] : Found installer for stream img1.gz roundrobin
[TRACE] : SWUPDATE running :  [notify_helper] : Round Robin Handler: No compiled-in config found, trying /etc/swupdate.handler.ini
[INFO ] : SWUPDATE running :  [notify_helper] : Round Robin Handler: Using '/dev/disk/by-partlabel/rootfs2' as target via 'raw' handler.
[TRACE] : SWUPDATE running :  [lua_dump_table] : properties subtype = image
[TRACE] : SWUPDATE running :  [lua_dump_table] : Inserting property subtype = image
[INFO ] : SWUPDATE running :  [notify_helper] : Round Robin Handler: Setting bootloader environment: ustate=1
[TRACE] : SWUPDATE running :  [l_handler_wrapper] : [Lua handler] returned: 0
[INFO ] : SWUPDATE successful ! SWUPDATE successful !
[TRACE] : SWUPDATE running :  [network_initializer] : Main thread sleep again !
[INFO ] : No SWUPDATE running :  Waiting for requests...
[INFO ] : SWUPDATE running :  [endupdate] : Swupdate was successful !

root@iot2050-debian:/media/usb0# reboot
         Stopping Sessio[  OK  ] Removed slice system-modprobe.slice.
[  OK  ] Removed slice system-swupdate\x2dusb.slice.
[  OK  ] Stopped target Bluetooth.
[  OK  ] Stopped target Graphical Interface.
[  OK  ] Stopped target Timers.
[  OK  ] Stopped Daily apt upgrade and clean activities.
[  OK  ] Stopped Daily apt download activities.
[  OK  ] Stopped Periodic ext4 Onli…ata Check for All Filesystems.
[  OK  ] Stopped Discard unused blocks once a week.
[  OK  ] Stopped Daily Cleanup of Temporary Directories.
[... 828 lines cut ...]
[  OK  ] Started Hostname Service.
[  OK  ] Started Modem Manager.
[  OK  ] Reached target Multi-User System.
         Starting watchdog daemon...
         Starting Network Manager Script Dispatcher Service...
[  OK  ] Started watchdog daemon.
[  OK  ] Reached target Graphical Interface.
         Starting Update UTMP about System Runlevel Changes...
[... cut ...]
[   22.096275] TI DP83867 b032400.mdio:01: attached PHY driver [TI DP83867] (mii_bus:phy_addr=b032400.mdio:01, irq=POLL)
[   22.107303] net eno1: started

Debian GNU/Linux 11 iot2050-debian ttyS3

iot2050-debian login: [   33.879859] tps62363-vout: disabling
[   36.441349] TI DP83867 b032400.mdio:01: Downshift occurred from negotiated speed 1Gbps to actual speed 100Mbps, check cabling!
[   36.452849] IPv6: ADDRCONF(NETDEV_CHANGE): eno1: link becomes ready
[   36.453846] icssg-prueth icssg0-eth eno1: Link is Up - 100Mbps/Full (downshifted) - flow control off

jan-kiszka, 2022-03-23:
Looks as if patching of the U-Boot env didn't work out on the first boot of the SWUpdate image. That patching will install the boot path switch. Could you share `fw_printent`and `journalctl -u patch-u-boot-env`?

amitsagargowda, 2022-03-23:
Please find o/p of both command below.

**fw_printenv o/p**
------------------
root@iot2050-debian:# fw_printenv
arch=arm
baudrate=115200
board=iot2050
board_a5e=A5E452229880AA04
board_name=IOT2050-ADVANCED
board_serial=M5BM8308
board_uuid=7FFD2E1C9B1C4BAC98DEE2B111EB527C
boot_a_script=load ${devtype} ${devnum}:${distro_bootpart} ${scriptaddr} ${prefix}${script}; source ${scriptaddr}
boot_efi_binary=load ${devtype} ${devnum}:${distro_bootpart} ${kernel_addr_r} efi/boot/bootaa64.efi; if fdt addr ${fdt_addr_r}; then bootefi ${kernel_addr_r} ${fdt_addr_r};else bootefi ${kernel_addr_r} ${fdtcontroladdr};fi
boot_efi_bootmgr=if fdt addr ${fdt_addr_r}; then bootefi bootmgr ${fdt_addr_r};else bootefi bootmgr;fi
boot_extlinux=sysboot ${devtype} ${devnum}:${distro_bootpart} any ${scriptaddr} ${prefix}${boot_syslinux_conf}
boot_fdt=try
boot_net_usb_start=usb start
boot_pci_enum=pci enum
boot_prefixes=/ /boot/
boot_script_dhcp=boot.scr.uimg
boot_scripts=boot.scr.uimg boot.scr
boot_syslinux_conf=extlinux/extlinux.conf
boot_targets=mmc1 mmc0 usb0 usb1 usb2
bootcmd=run distro_bootcmd
bootcmd_mmc0=devnum=0; run mmc_boot
bootcmd_mmc1=devnum=1; run mmc_boot
bootcmd_usb0=devnum=0; run usb_boot
bootcmd_usb1=devnum=1; run usb_boot
bootcmd_usb2=devnum=2; run usb_boot
bootdelay=2
bootm_size=0x10000000
cpu=armv8
distro_bootcmd=for target in ${boot_targets}; do run bootcmd_${target}; done
dtboaddr=0x89000000
efi_dtb_prefixes=/ /dtb/ /dtb/current/
eth1addr=e0:dc:a0:c4:91:df
eth2addr=e0:dc:a0:c4:91:de
fdt_addr_r=0x88000000
fdtaddr=0x88000000
fdtcontroladdr=fdf00480
fdtfile=ti/k3-am6548-iot2050-advanced.dtb
fw_version=2022.01-V01.02.01-78-gbf54a81
kernel_addr_r=0x82000000
load_efi_dtb=load ${devtype} ${devnum}:${distro_bootpart} ${fdt_addr_r} ${prefix}${efi_fdtfile}
loadaddr=0x82000000
mlfb=6ES7647-0BA00-1YA2
mmc_boot=if mmc dev ${devnum}; then devtype=mmc; run scan_dev_for_boot_part; fi
pxefile_addr_r=0x80100000
ramdisk_addr_r=0x88080000
rdaddr=0x88080000
scan_dev_for_boot=echo Scanning ${devtype} ${devnum}:${distro_bootpart}...; for prefix in ${boot_prefixes}; do run scan_dev_for_extlinux; run scan_dev_for_scripts; done;run scan_dev_for_efi;
scan_dev_for_boot_part=part list ${devtype} ${devnum} -bootable devplist; env exists devplist || setenv devplist 1; for distro_bootpart in ${devplist}; do if fstype ${devtype} ${devnum}:${distro_bootpart} bootfstype; then run scan_dev_for_boot; fi; done; setenv devplist
scan_dev_for_efi=setenv efi_fdtfile ${fdtfile}; for prefix in ${efi_dtb_prefixes}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${efi_fdtfile}; then run load_efi_dtb; fi;done;run boot_efi_bootmgr;if test -e ${devtype} ${devnum}:${distro_bootpart} efi/boot/bootaa64.efi; then echo Found EFI removable media binary efi/boot/bootaa64.efi; run boot_efi_binary; echo EFI LOAD FAILED: continuing...; fi; setenv efi_fdtfile
scan_dev_for_extlinux=if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${boot_syslinux_conf}; then echo Found ${prefix}${boot_syslinux_conf}; run boot_extlinux; echo SCRIPT FAILED: continuing...; fi
scan_dev_for_scripts=for script in ${boot_scripts}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${script}; then echo Found U-Boot script ${prefix}${script}; run boot_a_script; echo SCRIPT FAILED: continuing...; fi; done
scriptaddr=0x80000000
seboot_version=D01.02.02.02-0-g3af64a90-0x0000
soc=k3
usb_boot=usb start; if usb dev ${devnum}; then devtype=usb; run scan_dev_for_boot_part; fi
usb_pgood_delay=900
ustate=1
vendor=siemens
root@iot2050-debian:~#

**journalctl -u patch-u-boot-env**
---------------------------------
root@iot2050-debian:/tmp# journalctl -u patch-u-boot-env
-- Journal begins at Thu 2022-03-10 04:42:17 UTC, ends at Wed 2022-03-23 07:28:18 UTC. --
Mar 10 04:42:58 iot2050-debian systemd[1]: Starting update u-boot environment for swupdate...
Mar 10 04:42:59 iot2050-debian systemctl[454]: Removed /etc/systemd/system/multi-user.target.wants/patch-u-boot-env.service.
Mar 10 04:43:00 iot2050-debian systemd[1]: patch-u-boot-env.service: Succeeded.
Mar 10 04:43:00 iot2050-debian systemd[1]: Finished update u-boot environment for swupdate.
root@iot2050-debian:/tmp#

jan-kiszka, 2022-03-23:
Hmm, not sure yet what went wrong. The log says everything was fine, but the environment does contain what https://github.com/siemens/meta-iot2050/blob/master/recipes-bsp/patch-u-boot-env/files/patch-u-boot-env.config would bring in. Do you also get that variable state also when running `printenv` from the U-Boot shell? If so, please retry from Linux: `systemctl start patch-u-boot-env`.

gylstorffq, 2022-03-23:
`ustate` was introduced during the update by SWUpdate.

amitsagargowda, 2022-03-24:
@jan-kiszka  and @gylstorffq  I tried all steps mentioned above.

Actually I confirmed that uboot has variable ustate=0. Later i trigerred swupdate at end of successfull swupdate tranfer the ustate was set to 1 . I was expecting that rootfs to trigger rootfs change from rootfs1 to rootfs2 but i still see that it points to rootfs1 only. But now the ustate value is set to 3. I have attached complete log for your reference. What could go wrong here?  We are really stuck up here as this is one of the main part of our user story. Could you please assist us here.

iot2050-image-swu-example-iot2050-debian-iot2050.swu  tmp                                                      TMPDIR=/mnt/usb0/tmp/ swupdate -v -i ./iot2050-image-swu-example-iot2050-debian-iot2050.swudebian:/mnt/usb0# TMPDIR=/mnt/usb0/tmp/ swupdate -v -i ./iot2050-image-swu-example-iot2050-debian-iot2050.swu
Swupdate v2021.11.0

Licensed under GPLv2. See source distribution for detailed copyright notices.

[INFO ] : SWUPDATE running :  [lua_handlers_init] : Lua handler(s) found.
[INFO ] : SWUPDATE running :  [print_registered_handlers] : Registered handlers:
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     dummy
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     archive
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     tar
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     uboot
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     bootloader
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     diskformat
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     diskpart
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     uniqueuuid
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     lua
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     raw
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     rawfile
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     rawcopy
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     rdiff_image
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     rdiff_file
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     readback
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     shellscript
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     preinstall
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     postinstall
[INFO ] : SWUPDATE running :  [print_registered_handlers] :     roundrobin
[TRACE] : SWUPDATE running :  [listener_create] : got no socket at /mnt/usb0/tmp//swupdateprog from systemd
[TRACE] : SWUPDATE running :  [listener_create] : creating socket at /mnt/usb0/tmp//swupdateprog
[TRACE] : SWUPDATE running :  [network_initializer] : Main loop daemon
[TRACE] : SWUPDATE running :  [listener_create] : got no socket at /mnt/usb0/tmp//sockinstctrl from systemd
[TRACE] : SWUPDATE running :  [listener_create] : creating socket at /mnt/usb0/tmp//sockinstctrl
[TRACE] : SWUPDATE running :  [network_thread] : Incoming network request: processing...
[INFO ] : SWUPDATE started :  Software Update started !
[TRACE] : SWUPDATE running :  [network_initializer] : Software update started
[WARN ] : SWUPDATE running :  [swupdate_create_directory] : Directory scripts/ cannot be created due to : File exists
[WARN ] : SWUPDATE running :  [swupdate_create_directory] : Directory datadst/ cannot be created due to : File exists
[TRACE] : SWUPDATE running :  [extract_file_to_tmp] : Found file
[TRACE] : SWUPDATE running :  [extract_file_to_tmp] :   filename sw-description
[TRACE] : SWUPDATE running :  [extract_file_to_tmp] :   size 577
[TRACE] : SWUPDATE running :  [get_common_fields] : Version 0.1
[TRACE] : SWUPDATE running :  [add_properties] : Found properties for img1.gz:
[TRACE] : SWUPDATE running :  [add_properties_cb] :             Property subtype: image
[TRACE] : SWUPDATE running :  [_parse_images] : Found compressed Image: img1.gz in device : rootfs1,rootfs2 for handler roundrobin
[TRACE] : SWUPDATE running :  [extract_files] : Found file
[TRACE] : SWUPDATE running :  [extract_files] :         filename img1.gz
[TRACE] : SWUPDATE running :  [extract_files] :         size 869088818 required
[TRACE] : SWUPDATE running :  [extract_padding] : Expecting 336 padding bytes at end-of-file
[TRACE] : SWUPDATE running :  [network_initializer] : Valid image found: copying to FLASH
[INFO ] : SWUPDATE running :  Installation in progress
[TRACE] : SWUPDATE running :  [install_single_image] : Found installer for stream img1.gz roundrobin
[TRACE] : SWUPDATE running :  [notify_helper] : Round Robin Handler: No compiled-in config found, trying /etc/swupdate.handler.ini
[INFO ] : SWUPDATE running :  [notify_helper] : Round Robin Handler: Using '/dev/disk/by-partlabel/rootfs2' as target via 'raw' handler.
[TRACE] : SWUPDATE running :  [lua_dump_table] : properties subtype = image
[TRACE] : SWUPDATE running :  [lua_dump_table] : Inserting property subtype = image
[INFO ] : SWUPDATE running :  [notify_helper] : Round Robin Handler: Setting bootloader environment: ustate=1
[TRACE] : SWUPDATE running :  [l_handler_wrapper] : [Lua handler] returned: 0
[INFO ] : SWUPDATE successful ! SWUPDATE successful !
[TRACE] : SWUPDATE running :  [network_initializer] : Main thread sleep again !
[INFO ] : No SWUPDATE running :  Waiting for requests...
[INFO ] : SWUPDATE running :  [endupdate] : Swupdate was successful !
root@iot2050-debian:/mnt/usb0#
[... cut ...]
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
/dev/mmcblk0p3 on /data type ext4 (rw,noatime,nodiratime,nodioread_nolock,nodelalloc,commit=20,data=journal)
tmpfs on /run/user/0 type tmpfs (rw,nosuid,nodev,relatime,size=198056k,nr_inodes=49514,mode=700)
root@iot2050-debian:~#

amitsagargowda, 2022-03-25:
@jan-kiszka : We are using 16GB sd card as mass storage device. Yes currently both are plugged in and we prefer SD card out of it. Is there a way to make emmc not bootable?

jan-kiszka, 2022-03-25:
https://github.com/siemens/meta-iot2050/#selecting-a-boot-device

amitsagargowda, 2022-03-25:
@jan-kiszka : I removed sd card and tried on emmc the swupdate now works fine thanks a lot. But we need SD card approach also.
Currently my boot_targets is set to 
**boot_targets=mmc1 mmc0 usb0 usb1 usb2**
If i understand the boot order is emmc, sdcard and USBs right?

Is this creating the issue.

If i set **boot_targets=mmc0 mmc1 usb0 usb1 usb2**

Will it help 

What about setting **boot_targets=mmc0**

I would some thought from your side on this.

jan-kiszka, 2022-03-25:
Yes, mmc0 is SD, mmc1 is eMMC. If you move mmc0 to the front, you will only boot from eMMC if SD fails. If you drop everything but SD from boot_targets, only that will be tried for booting.

closed 2022-03-28 by amitsagargowda