siemens/meta-iot2050 #281: Access to GPIOs only with root privileges

bergmanu, 2022-03-15:
Access to the Arduino GPIOS via bash programms "mraa-gpio", "mraa-uart" and "mraa-i2c" is only working logged in as root or using sudo being logged in as user. 

Calling the programs as user results in "free(): invalid pointer" 
[image]

Is this intended? Is there any other way to access GPIOs as user without using sudo?

jan-kiszka, 2022-03-15:
You can try `chown`ing the sysfs nodes that have to be accessed and can be securely forwarded.

That `free()` bug is likely a error in some cleanup path. Needs to be analyzed.

jan-kiszka, 2022-03-15:
@fr0st61te, could you eventually have a look if / what goes wrong there in mraa when it calls that wrong free()? TIA.

bergmanu, 2022-03-15:
> You can try `chown`ing the sysfs nodes that have to be accessed and can be securely forwarded.
> 
> That `free()` bug is likely a error in some cleanup path. Needs to be analyzed.

Tried to `chown` the whole _/sys/class/gpio_ directory recursively, still same output

jan-kiszka, 2022-03-15:
Just recalling: We also use direct MMIO access for configuring the pins in mraa. That involves using /dev/mem - a no-go for anything unprivileged. So, this is doomed to fail.

fr0st61te, 2022-03-16:
> @fr0st61te, could you eventually have a look if / what goes wrong there in mraa when it calls that wrong free()? TIA.

Sure, I'll take a look!

jan-kiszka, 2023-06-19:
Merged into master, happily looking for more testers now!

closed 2023-06-19 by jan-kiszka