siemens/meta-iot2050 #282: Building a bullseye image

gxcora, 2022-03-17:
Hi:

1. We managed to build a buster image using `git checkout V01.01.01`. However, attempting to build an image using `V01.02.01` (which presumably is `bullseye`) results in the following error: `Can't find default configuration "arch/../configs/iot2050_defconfig"!`. Full error log attached. Commands we're running are as follows:
```    
git clone -b stable/V01.02  https://github.com/siemens/meta-iot2050
cd meta-iot2050/
git checkout V01.02.01
./kas-container --isar build kas-iot2050-example.yml
```

2.  We have some custom dpkgs and whl (python packages) files we need to be installed in the image. We could install the deb packages using the bb recipes, but not the whl files. What is your recommended approach to installing python packages, via whl files - eg; `pip3  install <package>.whl`?

3.  Also, we have some configurations we need to make. Most of these work using the file: `recipes-core/customizations-example/files/postinst`. However, some of our configurations don't work (such as `ufw enable`, or resetting the root password). What is your recommended approach to making these post installation configurations?

thanks

[log.do_dpkg_build.234338.txt](https://github.com/siemens/meta-iot2050/files/8281637/log.do_dpkg_build.234338.txt)

jan-kiszka, 2022-03-17:
1. The build recipes for the release V01.02.01 had a bug which caused this breakage. It was fixed via 5bf16e56d5999f794c8223f602d73fd3bff2109a in stable/V01.02. So build that branch head instead.
2. The simplest way to get a python package is indeed pulling it via pip - but it is not the cleanest one. Better is real packaging. See e.g. https://gitlab.com/cip-project/cip-core/isar-cip-core/-/commit/4d48bc4bf4feed3a2126dcbfc1740dc95e5c25bb where I just had to do that. What is your package? Any traces of it already on https://salsa.debian.org?
3. This depends on what you have to do. I'm not familiar with ufw, but I assume it is a systemd service that needs to be enabled so that it starts at every boot. If that is not done by default, a `systemctl enable ufw` (or so) in your customization postinst can be the trick. But I would consult the Debian package documentation on that first.
The root password can be easily set via `USER_root[password]` at image level, e.g. by setting this variable in your own image recipe, recipe append or an own configuration file (kas fragment). Check also if that `passwd --expire root` in the default postinst makes sense for you.

gxcora, 2022-03-18:
Jan:

thanks. I tried again as follows, but no luck. See attached log file. Not sure what I'm doing wrong.
```
git clone -b stable/V01.02  https://github.com/siemens/meta-iot2050
cd meta-iot2050/
./kas-container --isar build kas-iot2050-example.yml
```

[log.do_rootfs_install.446768.txt](https://github.com/siemens/meta-iot2050/files/8304045/log.do_rootfs_install.446768.txt)

jan-kiszka, 2022-03-18:
OK, stable lacks what was just merged into master: https://github.com/siemens/meta-iot2050/commit/1b2ee744451963384895626cc93b269e183aee17. I'll send a PR for that branch as well.

jan-kiszka, 2022-03-18:
#286

gxcora, 2022-03-20:
Thanks Jan:

I tried the latest `stable/V01.02`. It builds, but says:
```WARNING: iot2050-image-example-1.0-r0 do_image: This image is NOT COMPATIBLE with current official firmware releases!
It is recommended to build release V01.01.01 instead.
```

And it doesn't boot on the IOT2050. I guess that means I can't use `stable/V01.02` with bullseye yet?

These are the commands:
```
git clone -b stable/V01.02  https://github.com/siemens/meta-iot2050
cd meta-iot2050/
./kas-container --isar build kas-iot2050-example.yml
```

jan-kiszka, 2022-03-20:
Did you update the firmware?

gxcora, 2022-03-20:
Jan:

  ah no! How is this done?
  
  G

jan-kiszka, 2022-03-20:
Please see https://support.industry.siemens.com/tf/ww/en/posts/firmware-update-to-use-example-image-v1-2-1/275777/?page=0&pageSize=10

gxcora, 2022-03-21:
Thanks again Jan. All working well. I'll close this issue.

gxcora, 2022-03-30:
Jan:

I forgot about one more thing...as per my request for help above:
```
We have some custom dpkgs and whl (python packages) files we need to be installed in the image. We could install the deb packages using the bb recipes, but not the whl files.
```

The packages we have are simple Python scripts. I noted your suggestion above (ie; the `shtab` example). So I tried the following:
a. Copied the `shtab` folder as per https://gitlab.com/cip-project/cip-core/isar-cip-core/-/tree/master/recipes-python
b. Placed it in `recipes-app` . Changed the filename of the bitbake recipe to `shtab` 
c. Added `shtab` into `...github/meta-iot2050/recipes-core/images/iot2050-image-example.bb` in the `IMAGE_INSTALL += " \` section. 

This didn't work. See attached error file. Note sure what I'm doing wrong here. I've tried quite a few things, and none of them have worked.

Thanks again
[error_log_30Mar22.txt](https://github.com/siemens/meta-iot2050/files/8377298/error_log_30Mar22.txt)

jan-kiszka, 2022-03-30:
You probably need to declare one of the python deps from `DEBIAN_BUILD_DEPENDS` as `:native`. See https://gitlab.com/cip-project/cip-core/isar-cip-core/-/commit/14a5ec84e49fe4d3dfd22f9c6bb26d63075a497b. But double-check if those deps are actually needed or just copy&pasted from the example recipe.

jan-kiszka, 2023-06-19:
Probably resolved by now.

closed 2023-06-19 by jan-kiszka