siemens/meta-iot2050 #284: GPIO DIs only show 0 (Basic PG2)

bergmanu, 2022-03-17:
A customer in forum is describing that the DIs of the Arduino GPIOs are not working with IOT2050 Basic PG2- they only show 0, even if 24V is connected. 
More info see [here](https://support.industry.siemens.com/tf/ww/en/posts/trouble-reading-digital-input-pins/278302/?page=0&pageSize=10#post1075969)

I do not have a PG2 device at the moment. Could someone test that? 
Will check it with PG1 next week in office.

bergmanu, 2022-03-21:
I tested with IOT2050 PG1 in different cominations. GPIOs work on all of them! 

1. Firmware V1.0.0 with Example Image V1.1.1
2. Firmware V1.2.1 with Example Image V1.1.1
3. Firmware V1.2.1 with Example Image V1.2.1

Connected a DQ of a S7-1500 to DI0 of the Siemens Input/Output Module and checked its status via `mraa-gpio get 12` and the value changed between 0 and 1, depending whether the S7-1500 DQ is 0 or 1.

GQ411, 2022-03-24:
It's no occurred on my test environment.
Hardware: 
`iot2050 PG2 Basic + iot2000 i/o MODULER (6ES7647-0KA01-0AA2) FS 01`

software:
```
firmware V01.02.01_40d3fc0_20220106.
seboot_version: V01.02.01.04-0-g398B965C-0X000
```

Test:

1 connect with 24V to pin 12,  **mraa-gpio get 12**, it return **1**;
2 disconnect 24V form pin 12, **mraa-gpio get 12**, it return **0**;

it not reproduce on my environment.

chris0z, 2022-12-09:
Have the same issue 
IOT2050 Advanced FS:05 image V1.2.2
I changed the Jumper like described in the manual

Test:
1 connect with 24V to pin 12, mraa-gpio get 12, it return 0;
2 disconnect 24V form pin 12, mraa-gpio get 12, it return 0;

jan-kiszka, 2022-12-09:
I hope you didn't use 24V on the bare IO12 pin: without an I/O shield like https://mall.industry.siemens.com/mall/en/WW/Catalog/Product/6ES7647-0KA01-0AA2, those pins are 0..5V only (@bergmanu, correct me if I'm wrong).

chris0z, 2022-12-15:
Yes i used this 6ES7647-0KA01-0AA2 for my setup

chris0z, 2022-12-16:
You have to wire the specific GND pin to GND in my case M0 to GND. Than its working fine. Its not connected internally to the GND as expected. Hope that helps.

closed 2023-06-19 by jan-kiszka