siemens/meta-iot2050 #290: Build Fails on Ubuntu 20.04.4 LTS

anandcb123, 2022-03-22:
We have setup VM based on Ubuntu-20.04.4 LTS and trying to build iot-meta2050 , but build fails with error mentioned below.

We have installed docker.io which is one of the requirement. 

The log file is also attached for reference.

Are we missing something here?

[image]

[image]

[image]

[debootstrap.log](https://github.com/siemens/meta-iot2050/files/8320894/debootstrap.log)

[run.do_bootstrap.377.log](https://github.com/siemens/meta-iot2050/files/8320903/run.do_bootstrap.377.log)

jan-kiszka, 2022-03-22:
Same issue as #219?

jan-kiszka, 2023-06-19:
Probably solved by now