siemens/meta-iot2050 #291: Unable to load kas-isar:2.6.3 image

Girish-75020413, 2022-03-22:
Hello Team,

I am getting below error while executing the .gitlab-ci.yml (https://github.com/siemens/meta-iot2050/blob/master/.gitlab-ci.yml) file on my GitLab runner.

What changes should be done to resolve this error?

Running with gitlab-runner 14.8.[2]
  on i****s2* yqi[3](https://code.siemens.com/PI-DIGITALISATION/cc220iot2050imagesetup/-/jobs/71511115#L3)[4](https://code.siemens.com/PI-DIGITALISATION/cc220iot2050imagesetup/-/jobs/71511115#L4)8XN
  feature flags: FF_NETWORK_PER_BUILD:true
Preparing the "docker" executor
00:09
ERROR: The "ghcr.io/siemens/kas/kas-isar:2.[6](https://code.siemens.com/PI-DIGITALISATION/cc220iot2050imagesetup/-/jobs/71511115#L6).3" image is not present on list of allowed images:
- azul/*:*
- buildpack-deps:*
- circleci/node:*
- cirrusci/android-sdk:*
- cirrusci/flutter:*
- composer:*
- conanio/*:*
- gcc:*
- golang:*
- gradle:*

Regards,
Girish M

jan-kiszka, 2022-03-22:
Limitation of your local gitlab or, rather, of the its runner pool. It does not list kas-isar a permitted image. Moreover, it will likely also not support the privileged execution of those images which is needed for the build system. Clarify with your admins.

jan-kiszka, 2023-06-19:
Likely obsolete now.

closed 2023-06-19 by jan-kiszka