siemens/meta-iot2050 #298: Can build Image - debootstrap Segmentation fault (Tag: V01.02.01)

BenediktMiller, 2022-04-05:
Hello,

the build on the V01.02.01 tag fails with the following error:
(Command: `./kas-container build kas-iot2050-example.yml`)

```
I: Extracting liblzma5...
| I: Extracting zlib1g...
| W: Failure trying to run: chroot "/build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs" /sbin/ldconfig
| W: See /build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs/debootstrap/debootstrap.log for details
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-target.bb:do_bootstrap) failed with exit code '1'
```

The end of debootstrap.log looks like this:

```
--2022-04-05 20:59:41--  http://deb.debian.org/debian/pool/main/z/zlib/zlib1g_1.2.11.dfsg-2_arm64.deb
Resolving deb.debian.org (deb.debian.org)... 151.101.14.132, 2a04:4e42:3::644
Connecting to deb.debian.org (deb.debian.org)|151.101.14.132|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 87944 (86K) [application/x-debian-package]
Saving to: ‘/build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs//var/cache/apt/archives/partial/zlib1g_1%3a1.2.11.dfsg-2_arm64.deb’

     0K .......... .......... .......... .......... .......... 58% 3.73M 0s
    50K .......... .......... .......... .....                100% 9.36M=0.02s

2022-04-05 20:59:41 (4.98 MB/s) - ‘/build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs//var/cache/apt/archives/partial/zlib1g_1%3a1.2.11.dfsg-2_arm64.deb’ saved [87944/87944]

Segmentation fault (core dumped)
```

The host build system is ` Debian GNU/Linux 10 (buster)`

Is there any way to fix this issue?

jan-kiszka, 2022-04-06:
Is the host a VM or a native machine? How much RAM does it have?

Does it have qemu-user-static installed? If so, please remove that package, reboot and retry the build.

jan-kiszka, 2023-06-19:
Probably solved by now.

closed 2023-06-19 by jan-kiszka