siemens/meta-iot2050 #300: Can't boot the device with builded images, need to update the firmware?

yayitazale, 2022-04-11:
I have been able to successfully create sample images with different configurations (normal, LXDE, Coral...) but none of them boots on my IoT2050.

Do I have to create a "firmware image" first to update the firmware? How can I know if my device is PG1 or PG2? Where can I find the steps to update the firmware with the image?

Thanks

lifang0504, 2022-04-11:
Hi, you can refer to this below link:

https://support.industry.siemens.com/tf/ww/en/posts/firmware-update-to-use-example-image-v1-2-1/275777/?page=0&pageSize=10

jan-kiszka, 2022-04-11:
In addition, attaching a UART-serial converter to the X14 debug port can help a lot in debugging boot issues. The output will also dump the device type.

closed 2022-04-11 by yayitazale