siemens/meta-iot2050 #301: Unable to build custom image

Erokos, 2022-04-12:
Hi,
we're trying to build a custom version of our image using this repository and we're really having problems.
We started simple, our goal was to create an image without the nodered and mosquitto packages.
1. We copied the `recipes-core/images/iot2050-image-example.bb` to a new file `recipes-core/images/iot2050-image-my-example.bb`

2. Deleted the aforementioned packages from the `IOT2050_DEBIAN_DEBUG_PACKAGES` list in the file
3. Copied the `kas-iot2050-example.yml` to `kas-iot2050-my.yml` and set the `target: iot2050-image-my-example`
4. Modified the `Kconfig` file so that we can start the build from the menu.

We successfully built it, but when we try to boot it from an SD card or from emmc we get this error:

```
run bootcmd_mmc1
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Found U-Boot script /boot/boot.scr
532 bytes read in 2 ms (259.8 KiB/s)
## Executing script at 83000000
Loading /usr/lib/linux-image-5.10.104-cip3/...
** File not found /usr/lib/linux-image-5.10.104-cip3/ **
Loading /boot/vmlinux-5.10.104-cip3...
18298888 bytes read in 7220 ms (2.4 MiB/s)
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
SCRIPT FAILED: continuing...
```

Our package list in our recipe looks like this:
```
IOT2050_DEBIAN_DEBUG_PACKAGES = " \
    busybox \
    bash-completion \
    less \
    vim \
    psmisc \
    bsdmainutils \
    nano \
    ifupdown \
    iputils-ping \
    ssh \
    pciutils \
    usbutils \
    ethtool \
    rt-tests \
    stress-ng \
    build-essential \
    python3 \
    python-is-python3 \
    gawk \
    curl \
    wget \
    ca-certificates \
    resolvconf \
    python3-newt \
    gdb \
    gdbserver \
    gpiod \
    network-manager \
    modemmanager \
    ppp \
    isc-dhcp-client \
    cmake \
    autoconf \
    autotools-dev \
    default-jdk \
    teamd \
    rsyslog \
    net-tools \
    i2c-tools \
    sudo \
    "
```

Another thing we can't figure out is how to add packages to the image during the build process? For example git or docker? Can you give us some guidelines please? Any help would be much appreciated.

jan-kiszka, 2022-04-12:
Are you running firmware version 1.2.1?

jan-kiszka, 2022-04-12:
...and https://support.industry.siemens.com/tf/ww/en/posts/firmware-update-to-use-example-image-v1-2-1/275777/?page=0&pageSize=10

Erokos, 2022-04-12:
Hi,
thank you very much for your help the new firmware version did the trick and we successfully managed to use our custom image 😄 

So for the second part of the question, how can we add packages when building? E.g. docker and/or git?

jan-kiszka, 2022-04-12:
See the example I shared for a custom layer. It adds nginx. For more details on Isar, have look at https://github.com/ilbers/isar/blob/master/doc/user_manual.md.

Erokos, 2022-04-12:
Sorry to bother you, but I just can't seem to find the nginx example, can you please provide a link?

Erokos, 2022-04-12:
Yes, this did the trick and I managed to install git and htop but regarding docker it's abit trickier, especially since it's my first time dealing with this repo.

So, in my recipe that's basically a copy of the `iot2050-image-example.bb` I made the following changes:
```
IMAGE_PREINSTALL += " \
    ${IOT2050_DEBIAN_DEBUG_PACKAGES} \
    ${IOT2050_DEBIAN_WIFI_PACKAGES} \
    ${IOT2050_DEBIAN_BT_PACKAGES} \
    ${IOT2050_DEBIAN_ALSA_PACKAGES} \
    openvpn \
    git \
    htop \
    docker-ce \
    "

THIRD_PARTY_APT_KEYS_append = " https://download.docker.com/linux/debian/gpg;md5sum=1afae06b34a13c1b3d9cb61a26285a15"
DISTRO_APT_SOURCES_append = " conf/distro/docker-bullseye.list"
```

and added the follwing line to `isar/meta/conf/distro/debian-bullseye.list`:

```
deb [arch=arm64] https://download.docker.com/linux/debian	bullseye	stable
```

and the build complains:
```
 W: GPG error: https://download.docker.com/linux/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7EA0A9C3F273FCD8
| E: The repository 'https://download.docker.com/linux/debian bullseye InRelease' is not signed.
| WARNING: exit code 100 from a shell command.
```

Don't understand why apt doesn't see the key?

jan-kiszka, 2022-04-12:
Do you NEED docker-ce, or would docker.io (from Debian) be fine as well? Asking because the latter is much simpler to integrate, technically and legally (if you plan to redistribute).

If you need to pull from a 3rd-party repo, those two related variables need to go to a conf file, not just a recipe. That could be something similar to `conf/distro/iot2050-debian.conf`, or the local.conf generated by kas. Here is an old example (that we replaced with proper self-packaging / upstreaming): 1551a9abcae957620bc6765ae81840474a86a873

Erokos, 2022-04-13:
This example is very helpful, so if I understood correctly I'd have to do the same thing if I wanted to install `docker-ce`, right?
For `docker.io` I'm guessing it would be enough to add it to the `IMAGE_PREINSTALL` list?

One last question 🙂 - Currently debian bullseye is used as a base image, how would I set debian buster?

jan-kiszka, 2022-04-13:
Right, docker.io would be selected via IMAGE_PREINSTALL only - package is already in the Debian repo itself.

"Setting" buster is not that easy. If you go back in history of this repo, you will find the switching point where we moved to bullseye (around https://github.com/siemens/meta-iot2050/commit/8f6eae7e3a6206afcc22fbc3390298f7658ab337). Study what changed, specifically if you need any of the packages/recipes that we simplified on the switch to bullseye. If you need them, you'd have to roll those changes back. But details can be hairy, e.g. you may lose updates of those packages since the switch. Be warned.

Erokos, 2022-04-14:
Hi,
thanks to you I managed to build an image containing `docker.io` and one containing `docker-ce` along with some other packages. I realize that time is often a luxury we don't have but writing some documentation or examples like your commit regarding this repository would be very useful and probably less people would bother you - just a suggestion.

In any case, thank you kindly for all your support and very useful guidance, as far as I'm concerned, this issue is closed 🙂

closed 2022-04-14 by Erokos