siemens/meta-iot2050 #302: image build issue

applea9, 2022-04-13:
Hello,

I created custom image based on v1.2.2. but when I compile the image, I met such error

Configuration file '/etc/hosts'
 ==> File on system created by you or by a script.
 ==> File also in package provided by package maintainer.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** hosts (Y/I/N/O/D/Z) [default=N] ? dpkg: error processing package customizations-base (--configure):
 end of file on stdin at conffile prompt

The compilation host is Ubuntu 20.04, I checked the fail generated rootfs, I found /etc/hosts in already installed, but no the one created by customizations-base.

Could you help check the problem? thanks a lot.

jan-kiszka, 2022-04-13:
There can't be two packages owning the same file on the installed rootfs. If you want to set /etc/hosts yourself, replace or append to customization-base in your own layer.

applea9, 2022-04-13:
I don't have another package which owns /etc/hosts. It is only operated by costomizations-base package in meta-iot2050. It is fine when the base layer commit is v1.1.1. But when I switch to v1.2.2, this error occured.

jan-kiszka, 2022-04-13:
Were you building 883c59cbf5be87483810bb8798df6293f49e8a55 or stable/V01.02 HEAD (79f5701104a3f177af62606d8f1f41e78cc7044d)? The latter is required unless you use the package snapshot of V01.02.02 as well (`kas/opt/package-lock.yml`).

applea9, 2022-04-13:
I am using the commit of 883c59c, so if I build this commit, I need to use the package snapshot?

applea9, 2022-04-20:
It seems this issue is similar to #262 , and is solved by #263 . Can we merge #263 to stable/V01.02?

jan-kiszka, 2022-04-20:
Oh, forgot about that, indeed. Let's add it to the queue, but I'm not sure if we will have a V01.02.03.
@BaochengSu

BaochengSu, 2022-04-24:
@jan-kiszka I think it's OK to add it to the queue, but same as you I am not sure if V01.02.03 is needed, let's put it `undecide`.

closed 2022-05-05 by applea9