siemens/meta-iot2050 #303: ERROR: Did not find a cmdline Flattened Device Tree

OJthe123, 2022-04-14:
Hey.
I get this ERROR when booting from a just created SD card.

I used the method described here, install dockers, bmap-tools. 
The image was created without any errors.
But it cannot boot.

> Model: Siemens IOT2050 Advanced Base Board
DRAM:  2 GiB
MMC:   sdhci@4f80000: 1, sdhci@04FA0000: 0
Loading Environment from SPI Flash... SF: Detected w25q128 with page size 256 By    tes, erase size 64 KiB, total 16 MiB
OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: pruss0_eth
Hit any key to stop autoboot:  0
====== uuid conflict check begin ======
switch to partitions #0, OK
mmc1(part 0) is current device
switch to partitions #0, OK
mmc0 is current device
resetting USB...
USB0:   Register 2001040 NbrPorts 2
Starting the controller
USB XHCI 1.10
USB1:   Register 2001040 NbrPorts 2
Starting the controller
USB XHCI 1.10
scanning bus 0 for devices... 3 USB Device(s) found
scanning bus 1 for devices... 1 USB Device(s) found
scanning usb for storage devices... 0 Storage Device(s) found
Device 0: unknown device
Device 1: unknown device
Device 2: unknown device
====== uuid conflict check end ======
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
704 bytes read in 1 ms (687.5 KiB/s)
> Executing script at 83000000
Loading /usr/lib/linux-image-5.10.64/...
** File not found /usr/lib/linux-image-5.10.64/ **
Loading /boot/vmlinux-5.10.64...
18296840 bytes read in 4050 ms (4.3 MiB/s)
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
SCRIPT FAILED: continuing...
=>

OJthe123, 2022-04-14:
Thanks for your help.

I build the image right here from this Github project on my Linux Mint 19.1 computer.

The full log is this

```
SIMATIC IOT2050 SE-Boot Version: D01.00.00.09-0-g9827cee9-0x0000
BuildDate: 20200317
SYSFW ABI: 2.9 [version: 19] [19.12.1-v2019.12a (Terrific Lla]
AVS@[1100 1200 1200]
=======================
Board: IOT2050-ADVANCED
Serial: N9BV6121
MLFB: 6ES7647-0BA00-1YA2
UUID: 4D241DF200E14228BE2964E3424C867B
A5E: A5E452229880AA05
MAC[0]: 8c-f3-19-3a-0a-df
MAC[1]: 8c-f3-19-3a-0a-e0
Initlialzing DDR ...done.
SKU: SE
Loading PK... ok
PK count: 00
PK version: 00
SV: 00-00
Security ID 0x5125e53f-0x842e215d
Security policy: soft
Loading image...
Loading image...
Loading image...
Loading image...
NOTICE:  BL31: v2.1(release):v2.1-421-g996d37930-dirty
NOTICE:  BL31: Built : 20:04:56, Mar 18 2020
I/TC:
I/TC: OP-TEE version: 3.6.0-16-ge260ea8d-dev (gcc version 8.3.0 (Debian 8.3.0-2)                  ) #1 Wed Mar 18 20:08:49 UTC 2020 aarch64
I/TC: Initialized

U-Boot SPL 2019.01-V01.00.00.2-0-g8e86139 (Mar 18 2020 - 20:04:56 +0000)
Trying to boot from SPI
Using 'conf-iot2050-advanced' configuration
## Verifying Hash Integrity ... sha256,rsa4096:custMpk+ OK
## Checking hash(es) for Image u-boot ... sha256+ OK
## Checking hash(es) for Image iot2050-advanced.dtb ... sha256+ OK

U-Boot 2019.01-V01.00.00.2-0-g8e86139 (Mar 18 2020 - 20:04:56 +0000)

Model: Siemens IOT2050 Advanced Base Board
DRAM:  2 GiB
MMC:   sdhci@4f80000: 1, sdhci@04FA0000: 0
Loading Environment from SPI Flash... SF: Detected w25q128 with page size 256 By                  tes, erase size 64 KiB, total 16 MiB
OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: pruss0_eth
Hit any key to stop autoboot:  0
====== uuid conflict check begin ======
switch to partitions #0, OK
mmc1(part 0) is current device
switch to partitions #0, OK
mmc0 is current device
resetting USB...
USB0:   Register 2001040 NbrPorts 2
Starting the controller
USB XHCI 1.10
USB1:   Register 2001040 NbrPorts 2
Starting the controller
USB XHCI 1.10
scanning bus 0 for devices... 3 USB Device(s) found
scanning bus 1 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found

Device 0: unknown device

Device 1: unknown device

Device 2: unknown device
====== uuid conflict check end ======
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
704 bytes read in 1 ms (687.5 KiB/s)
## Executing script at 83000000
Loading /usr/lib/linux-image-5.10.64/...
** File not found /usr/lib/linux-image-5.10.64/ **
Loading /boot/vmlinux-5.10.64...
18296840 bytes read in 4051 ms (4.3 MiB/s)
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
SCRIPT FAILED: continuing...
=> 
```

jan-kiszka, 2022-04-14:
Ah, actually missed that your first log already contained the hint: This is the old firmware version which no longer works with the latest image release and later. Please update to V01.02.01 (see https://support.industry.siemens.com/tf/ww/en/posts/firmware-update-to-use-example-image-v1-2-1/275777/?page=0&pageSize=10).

jan-kiszka, 2023-07-19:
This was likely resolved by now.

closed 2023-07-19 by jan-kiszka