siemens/meta-iot2050 #310: IOT2050 - 4G - Network error : Socket hang up

ikramid, 2022-05-02:
Describing the bug :
Using a 4G mPCIe card (TELIT LE910CA-EU) on IOT2050, the onboarding was successful but when trying to push some data, the following error occured :
Error: Network error occured Network error occured request to https://southgate.eu1.mindsphere.io/api/agentmanagement/v3/agents/567d2fb9737848fe9743cc4f682c5efd/dataSourceConfiguration failed, reason: socket hang up

Additional context
The internet is working very well with other nodes on Node-red, but its speed is low : approx. 9Mbit/s for download and 0.4Mbit/s for upload (the speed test was done using speedtest-cli)
The same flow was tested using a wired network and it worked pretty well.

Node-RED version: v1.2.7
MindConnect Node-RED Agent: v3.12.0

jan-kiszka, 2022-05-02:
@sn0wcat, could we see such errors on rather slow links?

sn0wcat, 2022-05-02:
@jan-kiszka the error is coming from nodejs, an it happens in the TCP stack, so yes you could see this on the very slow links, but there is nothing I can do about it.

jan-kiszka, 2023-07-19:
No follow-ups since then, so likely resolved.

closed 2023-07-19 by jan-kiszka