siemens/meta-iot2050 #311: 4G card - IOT2050 - connection lost

ikramid, 2022-05-03:
Hello,

I aim to use a 4G card (Telit LE910C1-EU) with my IOT2050, but as soon as I plug it, I lost the connection on Putty.
I don't think it is a driver issue, but don't know what could it be.

Do you have any idea about the source of the problem ?

Thanks.

lifang0504, 2022-05-06:
Hello, do you have any mpcie pin definition document about this 4G card  (Telit LE910C1-EU)  ? We met this problem before when using ZTE 4G ME3630 , we need to double check if it is the same problem.

thank you so much .

ikramid, 2022-05-06:
Hello,

I have the document attached. I think that it's what you need.
If it's not, please tell me to find the right one.

Many thanks in advance.

lifang0504, 2022-05-07:
Hello, did you send the document via email ? But can't find this document, so can you send it again. 
thank you so much!

ikramid, 2022-05-09:
[Telit_LE910Cx_Interface_Board_Hardware_Design_Guide_r4.pdf](https://github.com/siemens/meta-iot2050/files/8652533/Telit_LE910Cx_Interface_Board_Hardware_Design_Guide_r4.pdf)
Hello,
Here's the attached file.

Thank you

lifang0504, 2022-05-10:
Hello,
Have received. I will ask HW colleagues to check that, if get any response will reply asap.
thank you so much!

lifang0504, 2022-05-10:
Hello,

What's pity, this is not we want. 
We want the **minipcie pin definition** of this 4G module(Telit LE910C1-EU), same as below. Can you help to provide ? Thank you so much!
[image]

ikramid, 2022-06-03:
Hello,

Unfortunately, it is shown in any picture, but according to technical support of the card manufacturer, the card  follows the standard mPCIe layout of the pins. So odd numbers on the top side and even numbers on the bottom side.

We had an old version of the same card which work properly, but the recent version isn't.
The difference between the two versions is a pinout change. Please find the change on the attached picture.
[image]

Do you think that it might me the problem ?

Thank you .

lifang0504, 2022-06-06:
Can you give me all mPCIe pin definition not just the pin 47 and pin 49 of this Telit LE910C1-EU? Due to different card maybe have different pin definition . Because we didn't have this card so our hardware colleagues need the whole pin definition to do research on this problem .
**such as below picture, the pin definition from pin1 to pin52.**
[image]
[image]

can you provide this information ? thank you so much!

BaochengSu, 2022-06-06:
Please make sure the new revision of the module is using PCIE signal, not the USB3.0 signal. We noticed there were some modules using usb3.0 by default, see issue #109

jan-kiszka, 2023-07-19:
I suppose this is resolved.

closed 2023-07-19 by jan-kiszka