siemens/meta-iot2050 #312: NixOS?

mmport80, 2022-05-03:
I would like to install NixOS. Any tips on where to start?

michaeladler, 2022-05-03:
Hi @mmport80 ,

I guess a good starting point would be the [NixOS wiki page for ARM](https://nixos.wiki/wiki/NixOS_on_ARM) (in particular the section "Porting NixOS to new boards").

The next step is to translate the relevant ISAR recipes from meta-iot2050 to corresponding Nix expressions. In particular, you might be interested in the [kernel config and patches](https://github.com/siemens/meta-iot2050/tree/master/recipes-kernel/linux) and [board-support packages](https://github.com/siemens/meta-iot2050/tree/master/recipes-bsp), e.g. firmware.

I hope that gives you a starting point.

mmport80, 2022-05-03:
Hmm, ISAR seems like it has some NixOS features, but for Debian. Learnt something new. Thanks! : )

closed 2022-05-03 by mmport80

michaeladler, 2022-05-03:
Yes, although there is a lot more mutable state involved. Ideally, the mutable state is kept to a minimum though. Good luck and feel free to share your results, maybe someone else will find it useful, too :)