siemens/meta-iot2050 #315: Reading digital input with node-red-intel-gpio

wjomo, 2022-05-13:
Hello,
I am using a IOT2050 with the Example Image V1.2.2. On this system I am using node red and the Siemens IO Shield for reading some digital inputs. The digital input node reads the status of the signal, if its changing the node will send a message. Thank works fine.
I have checked the option **Send initial message with level of pin**, because I would like to read the digital input when the box is starting or node red is restarting. But at startup there is no message sent from the node, only when I deploy my flow there is a message sent. I thinks it is very important to read the status at startup or restart the flow, otherwise there will be a undefined status.

I am thinking about two useful features for that node:

1. Send the status of the pin at startup or restart the flow.
2. Reading the input periodically with a scaleable cycle-time.

But I have no idea if that features are possible to implement.

I would be happy if that were possible.

Thanks for your help an have an nice day.

fr0st61te, 2022-05-16:
1. Once you'll deploy the flow, you'll see that will be started at startup or after Node-Red restart.
2. Probably you need 'inject' functions from Node-Red nodes, as example https://theautomationplace.com/2019/10/30/time-based-automations-with-node-red/

wjomo, 2022-05-17:
Hi,
thanks for your answer.
I am using the pre installed nodes on the Example Image V1.2.2 for reading the digital inputs with node red.
At my use-case I tested the behaviour of the digital input nodes with checking the checkbox **Send initial message with level of pin** and without checking these option. With checking that option the node only send an status of the pin when I deploy the flow. At startup or restart Node-Red nothing happens.

I can t work with inject nodes, becuase there is no possibility to connect the digital input node with an inject node.

It would be great, if there is a way to adjust a time for a cyclic reading the input as it works with the analog input node. There is an option for adjusting the cycle time for reading the value.

I don t know what could I do to solve that problem.

Thanks for your help.

wjomo, 2022-05-29:
@fr0st61te 
Hi Ivan, 
do you have some further information for me about my issue?

Thank you an have an nice day.

fr0st61te, 2022-05-29:
> @fr0st61te Hi Ivan, do you have some further information for me about my issue?

@wjomo there is issue as you stated about messages on startup from DIN node, trying to figure out how to solve it.

wjomo, 2022-05-30:
Hi Ivan,
thanks for your quick answer.
If I understand correctly, you are trying to find a solution for that problem? Please let me know, if I can support you.

Thanks

fr0st61te, 2022-05-30:
> Hi Ivan, thanks for your quick answer. If I understand correctly, you are trying to find a solution for that problem? Please let me know, if I can support you.
> 
> Thanks

I'm testing the solution, I hope I'll get back with it at end of the week.

fr0st61te, 2022-05-31:
@wjomo could you please check on your side If I provide you js/html file for DIN node?

fr0st61te, 2022-06-01:
> Hi Ivan, I would like to test it on my iot-controller if you provide the new file to me. Can you give me a little instruction how I have to use new js/html file for DIN node (in which file do I have to replace or change the js/html file?)
> 

1. make the backup of  mraa-gpio-din.js and mraa-gpio-din.html at /usr/lib/node_modules/node-red/node_modules/node-red-node-intel-gpio.
2. change on these files:
https://pastebin.com/VAAR1E3b - mraa-gpio-din.js
https://pastebin.com/UEbK7JWe - mraa-gpio-din.html
3. `systemctl restart node-red`, now there is input connection for DIN node, you can connect inject node with DIN and control when you want to report about DIN status.

wjomo, 2022-06-03:
HI Ivan,
thanks for your discription.

Now I have tested the new version of the DIN node on my controller with example Image V1.2.2.. The node work as expected. You did a great job. 👍 

I have wired an inject node at the new DIN input. That inject node send a message at startup. The DIN node send a message everytime when it is triggered by the inject node.
It works at start/stop node red service and by deploying the nodes.

Without triggering the DIN node it send a message by changing the input state.

Will that change of the DIN node be available at the example image V1.2.2 in future?

Thanks a lot for your help and enjoy the weekend.

Best regards.

jan-kiszka, 2022-06-07:
I'm not aware of any plans for a 1.2.3, so the fix should become available with the next major release.

wjomo, 2022-06-07:
@fr0st61te  Hi Ivan, please excuse my newbie question, what does "PR under review" mean?

Thanks

jan-kiszka, 2023-06-19:
Fixed via upstream mraa - IIRC.

closed 2023-06-19 by jan-kiszka