siemens/meta-iot2050 #316: preinstall local node-red node to image

applea9, 2022-05-17:
Hi,

I have created a node-red node locally, which is not published in npm registry. I would like to preinstall this node in the image, could you give some hint to to achieve it?

Thanks.

jan-kiszka, 2022-05-18:
Have a look at https://github.com/siemens/meta-iot2050/tree/master/recipes-app/node-red-gpio. Your recipe could be fairly similar, except for where the node sources would come from (maybe in-tree, file://, maybe from another local repo).

applea9, 2022-05-19:
Thanks for your answer.

When I install npm package with npm.bbclass, I want to add extra custom arguments in addition to the predefined npm arguments. After checking the npm.bbclass file, I didn't find such options. Is it possible to do it?

jan-kiszka, 2022-05-19:
npm.bbclass is for pulling pre-packaged nodes from npmjs.com. That's not suited here and therefore also not used by the example I cited.

applea9, 2022-05-19:
Ok, Thanks a lot.

closed 2022-05-19 by applea9