siemens/meta-iot2050 #337: Unable to Build kas-iot2050-example.yml

Girish-75020413, 2022-07-19:
Hello Team,

We have set up a new Linux machine with below configuration on which we are running meta-iot2050 pipelines. The image generation process is failing with errors and below is the complete log. Please verify and let us know what needs to be done to resolve this.

$ ./kas-container --isar build kas-iot2050-example.yml
2022-07-19 10:10:21 - INFO     - kas 2.6.3 started
2022-07-19 10:10:21 - INFO     - /repo$ git rev-parse --show-toplevel
2022-07-19 10:10:21 - INFO     - /repo$ git rev-parse --show-toplevel
2022-07-19 10:10:21 - INFO     - /repo$ git rev-parse --show-toplevel
2022-07-19 10:10:21 - INFO     - Using /repo as root for repository meta-iot2050
2022-07-19 10:10:21 - INFO     - /work$ git clone -q https://github.com/ilbers/isar /work/isar
2022-07-19 10:10:21 - INFO     - /work$ git clone -q https://github.com/siemens/meta-coral /work/meta-coral
2022-07-19 10:10:21 - INFO     - /work$ git clone -q https://gitlab.com/cip-project/cip-core/isar-cip-core.git /work/cip-core
2022-07-19 10:10:25 - INFO     - Repository meta-coral cloned
2022-07-19 10:10:26 - INFO     - Repository cip-core cloned
2022-07-19 10:10:34 - INFO     - Repository isar cloned
2022-07-19 10:10:34 - INFO     - /repo$ git rev-parse --show-toplevel
2022-07-19 10:10:34 - INFO     - Using /repo as root for repository meta-iot2050
2022-07-19 10:10:34 - INFO     - /work/isar$ git status -s
2022-07-19 10:10:34 - INFO     - /work/isar$ git rev-parse --verify -q origin/cf3f18a649ff42726677a58b224b9bb58ea4ba46
2022-07-19 10:10:34 - INFO     - /work/isar$ git checkout -q cf3f18a649ff42726677a58b224b9bb58ea4ba46
2022-07-19 10:10:34 - INFO     - /work/meta-coral$ git status -s
2022-07-19 10:10:34 - INFO     - /work/meta-coral$ git rev-parse --verify -q origin/2e6775eb77d82fb6d0aef45cae73c4738926eaad
2022-07-19 10:10:34 - INFO     - /work/meta-coral$ git checkout -q 2e6775eb77d82fb6d0aef45cae73c4738926eaad
2022-07-19 10:10:34 - INFO     - /work/cip-core$ git status -s
2022-07-19 10:10:34 - INFO     - /work/cip-core$ git rev-parse --verify -q origin/ed318f247beef10e0433c6ed699b4c1ab008fa6a
2022-07-19 10:10:34 - INFO     - /work/cip-core$ git checkout -q ed318f247beef10e0433c6ed699b4c1ab008fa6a
2022-07-19 10:10:34 - INFO     - /repo$ git rev-parse --show-toplevel
2022-07-19 10:10:34 - INFO     - Using /repo as root for repository meta-iot2050
2022-07-19 10:10:34 - INFO     - /repo$ git rev-parse --show-toplevel
2022-07-19 10:10:34 - INFO     - Using /repo as root for repository meta-iot2050
2022-07-19 10:10:34 - INFO     - /work/isar$ /tmp/tmplsnm2o_9/get_bb_env /build
2022-07-19 10:10:34 - INFO     - /repo$ git rev-parse --show-toplevel
2022-07-19 10:10:34 - INFO     - Using /repo as root for repository meta-iot2050
2022-07-19 10:10:34 - INFO     - /build$ /work/isar/bitbake/bin/bitbake -c build iot2050-image-example
2022-07-19 10:10:37 - INFO     - Parsing recipes...done.
2022-07-19 10:10:38 - INFO     - Parsing of 112 .bb files complete (0 cached, 112 parsed). 112 targets, 0 skipped, 0 masked, 0 errors.
2022-07-19 10:10:38 - INFO     - NOTE: Resolving any missing task queue dependencies
2022-07-19 10:10:43 - INFO     - Initialising tasks...done.
2022-07-19 10:10:43 - INFO     - Sstate summary: Wanted 50 Local 0 Network 0 Missed 50 Current 0 (0% match, 0% complete)
2022-07-19 10:10:43 - INFO     - NOTE: Executing Tasks
2022-07-19 10:10:43 - INFO     - NOTE: Running task 1 of 565 (/work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-target.bb:do_fetch)
2022-07-19 10:10:43 - INFO     - NOTE: Running task 2 of 565 (/work/isar/meta/recipes-devtools/isar-apt/isar-apt.bb:do_fetch)
2022-07-19 10:10:43 - INFO     - NOTE: Running task 3 of 565 (/work/isar/meta/recipes-devtools/base-apt/base-apt.bb:do_fetch)
2022-07-19 10:10:43 - INFO     - NOTE: Running task 4 of 565 (/work/isar/meta/recipes-devtools/buildchroot/buildchroot-target.bb:do_fetch)
2022-07-19 10:10:43 - INFO     - NOTE: recipe isar-apt-1.0-r0: task do_fetch: Started
2022-07-19 10:10:43 - INFO     - NOTE: recipe isar-apt-1.0-r0: task do_fetch: Succeeded
2022-07-19 10:10:43 - INFO     - NOTE: Running task 5 of 565 (/work/isar/meta/recipes-devtools/isar-apt/isar-apt.bb:do_unpack)
2022-07-19 10:10:43 - INFO     - NOTE: recipe base-apt-1.0-r0: task do_fetch: Started
2022-07-19 10:10:43 - INFO     - NOTE: recipe isar-apt-1.0-r0: task do_unpack: Started
2022-07-19 10:10:43 - INFO     - NOTE: recipe isar-bootstrap-target-1.0-r0: task do_fetch: Started
2022-07-19 10:10:43 - INFO     - NOTE: recipe base-apt-1.0-r0: task do_fetch: Succeeded
2022-07-19 10:10:43 - INFO     - NOTE: Running task 6 of 565 (/work/isar/meta/recipes-devtools/base-apt/base-apt.bb:do_unpack)
2022-07-19 10:10:43 - INFO     - NOTE: recipe isar-bootstrap-target-1.0-r0: task do_fetch: Succeeded
2022-07-19 10:10:43 - INFO     - NOTE: Running task 7 of 565 (/work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-target.bb:do_unpack)
2022-07-19 10:10:43 - INFO     - NOTE: recipe isar-apt-1.0-r0: task do_unpack: Succeeded
2022-07-19 10:10:43 - INFO     - NOTE: Running task 8 of 565 (/work/isar/meta/recipes-devtools/isar-apt/isar-apt.bb:do_cache_config)
2022-07-19 10:10:43 - INFO     - NOTE: recipe buildchroot-target-1.0-r0: task do_fetch: Started
2022-07-19 10:10:43 - INFO     - NOTE: recipe buildchroot-target-1.0-r0: task do_fetch: Succeeded
2022-07-19 10:10:43 - INFO     - NOTE: recipe base-apt-1.0-r0: task do_unpack: Started
2022-07-19 10:10:43 - INFO     - NOTE: Running task 9 of 565 (/work/isar/meta/recipes-devtools/buildchroot/buildchroot-target.bb:do_unpack)
2022-07-19 10:10:43 - INFO     - NOTE: recipe base-apt-1.0-r0: task do_unpack: Succeeded
2022-07-19 10:10:43 - INFO     - NOTE: Running task 10 of 565 (/work/isar/meta/recipes-devtools/base-apt/base-apt.bb:do_cache)
2022-07-19 10:10:43 - INFO     - NOTE: recipe isar-apt-1.0-r0: task do_cache_config: Started
2022-07-19 10:10:43 - INFO     - NOTE: recipe isar-bootstrap-target-1.0-r0: task do_unpack: Started
2022-07-19 10:10:43 - INFO     - NOTE: recipe base-apt-1.0-r0: task do_cache: Started
2022-07-19 10:10:43 - INFO     - NOTE: recipe base-apt-1.0-r0: task do_cache: Succeeded
2022-07-19 10:10:43 - INFO     - NOTE: Running task 11 of 565 (/work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-host.bb:do_fetch)
2022-07-19 10:10:43 - INFO     - NOTE: recipe buildchroot-target-1.0-r0: task do_unpack: Started
2022-07-19 10:10:43 - INFO     - NOTE: recipe isar-bootstrap-target-1.0-r0: task do_unpack: Succeeded
[... 80 lines cut ...]
2022-07-19 10:10:45 - ERROR    - ERROR: Task (/work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-host.bb:do_bootstrap) failed with exit code '1'
2022-07-19 10:10:46 - INFO     - NOTE: recipe k3-rti-wdt-1.1-r0: task do_fetch: Succeeded
2022-07-19 10:10:51 - INFO     - NOTE: recipe trusted-firmware-a-iot2050-2.6-r0: task do_fetch: Succeeded
2022-07-19 10:11:04 - INFO     - NOTE: recipe ti-pruss-firmware-08.00.00.004-r0: task do_fetch: Succeeded
2022-07-19 10:11:05 - INFO     - NOTE: Tasks Summary: Attempted 27 tasks of which 0 didn't need to be rerun and 2 failed.
2022-07-19 10:11:05 - INFO     - 
2022-07-19 10:11:05 - INFO     - Summary: 2 tasks failed:
2022-07-19 10:11:05 - INFO     - /work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-target.bb:do_bootstrap
2022-07-19 10:11:05 - INFO     - /work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-host.bb:do_bootstrap
2022-07-19 10:11:05 - INFO     - Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
2022-07-19 10:11:05 - ERROR    - Command "/build$ /work/isar/bitbake/bin/bitbake -c build iot2050-image-example" failed
--- Error summary ---
ERROR: isar-bootstrap-target-1.0-r0 do_bootstrap: Execution of '/build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/temp/run.do_bootstrap.414' failed with exit code 1:
sudo: effective uid is not 0, is /usr/bin/sudo on a file system with the 'nosuid' option set or an NFS file system without root privileges?
WARNING: exit code 1 from a shell command.
ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/temp/log.do_bootstrap.414
ERROR: Task (/work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-target.bb:do_bootstrap) failed with exit code '1'
ERROR: isar-bootstrap-host-1.0-r0 do_bootstrap: Execution of '/build/tmp/work/iot2050-debian-arm64/isar-bootstrap-host/1.0-r0/temp/run.do_bootstrap.420' failed with exit code 1:
sudo: effective uid is not 0, is /usr/bin/sudo on a file system with the 'nosuid' option set or an NFS file system without root privileges?
WARNING: exit code 1 from a shell command.
ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/isar-bootstrap-host/1.0-r0/temp/log.do_bootstrap.420
ERROR: Task (/work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-host.bb:do_bootstrap) failed with exit code '1'
Cleaning up project directory and file based variables
00:00
ERROR: Job failed: exit status 1

jan-kiszka, 2022-07-19:
"sudo: effective uid is not 0, is /usr/bin/sudo on a file system with the 'nosuid' option set or an NFS file system without root privileges?"

What is your host system? What is the file system you are running the kas-container from? Does it have that "nosuid" set by chance?

Girish-75020413, 2022-07-19:
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.3 LTS
Release:        20.04
Codename:       focal

/dev/mapper/ubuntu--vg-lv--0 ext4       80G   15G   62G  19% /
/dev/mapper/ubuntu--vg-home  ext4       20G  5.2G   14G  28% /home
/dev/sda2                    ext4      974M  215M  692M  24% /boot
/dev/mapper/ubuntu--vg-tmp   ext4       20G  128K   19G   1% /tmp
/dev/mapper/ubuntu--vg-var   ext4       20G  6.5G   13G  35% /var
/dev/loop3                   squashfs   68M   68M     0 100% /snap/lxd/22753
/dev/loop2                   squashfs   62M   62M     0 100% /snap/core20/1270
/dev/loop0                   squashfs   47M   47M     0 100% /snap/snapd/16292
/dev/loop1                   squashfs   68M   68M     0 100% /snap/lxd/21835
/dev/loop7                   squashfs   56M   56M     0 100% /snap/core18/2409
/dev/loop4                   squashfs   62M   62M     0 100% /snap/core20/1518
/dev/loop5                   squashfs   47M   47M     0 100% /snap/snapd/16010
/dev/loop8                   squashfs   56M   56M     0 100% /snap/core18/2538

BaochengSu, 2022-09-02:
Better to have the disk requirement recorded into the README.

We can also provide a typical time-space usage of the building such as on a 8-core/16G system how many time is taken to build the example image, and how big the storage is consumed.

jan-kiszka, 2023-06-19:
I suppose this can be closed by now.

closed 2023-06-19 by jan-kiszka