siemens/meta-iot2050 #339: Hardware watchdog not working

Vishal8523, 2022-07-27:
The hardware watchdog configured in IOT2050 for the kernel 5.10.100 has issues, the watchdog does not reboot
whereas, the watchdog configured in IOT2050 for the kernel 4.19.94 works fine with the same command.

Command used: **touch /dev/watchdog**

HW version of IOT2050 gateway: PG1 with latest bootloader 1.02.01

jan-kiszka, 2022-07-27:
Which image version is affected here? Did you try with the latest release? With latest one built from master? Could you share a complete boot log from power-on?

Note that latest firmware version is 1.2.1 (https://support.industry.siemens.com/cs/document/109741799/downloads-for-simatic-iot20x0?dti=0&lc=en-WW).

Vishal8523, 2022-07-27:
Logs sent to your e-mail

jan-kiszka, 2022-10-18:
As reported via email: Local configuration changes broke something. If you can reproduce the issue with the reference swupdate image, please reopen.

closed 2022-10-18 by jan-kiszka