siemens/meta-iot2050 #356: Build V01.03.01 isar Image error

xiaoma0216, 2022-09-19:
Hi, I am trying to build isar image with the newest release version 01.03.01, but I had met some issues when compling it. 
the compling system will stuck on the following screen for a long time (might be tens of minutes)
[image]

then it would reports these "2 tasks failed", any clues to solve it? thanks a lot.

BaochengSu, 2022-09-19:
Hi, this is an known issue #351 , we also noticed the exactly same issue on the github action. The root cause is that the tag `debian/0.15.0-1` is removed from upstream repo.

This issue has been addressed by #352, so you can use the master branch. If you have to use the 1.3.1 tag then please use my branch [su/ci-fix-1.3.1](https://github.com/siemens/meta-iot2050/tree/su/ci-fix-1.3.1) to retry the building,

BaochengSu, 2022-09-19:
Or you can use [stable/V01.03](https://github.com/siemens/meta-iot2050/tree/stable/V01.03) branch because it will not be subject to change in the future.

closed 2022-09-19 by xiaoma0216