siemens/meta-iot2050 #357: ufw ipv6 support

STD-FP, 2022-09-19:
I tried to install ufw on the image created with version V01.03.01

when you enable the firewall with
`sudo ufw enable`

```
ERROR: problem running ufw-init
ip6tables-restore v1.8.7 (nf_tables): Couldn't load match `rt ': No such file or directory

Error occurred at line: 24
Try `ip6tables-restore -h 'or' ip6tables-restore --help 'for more information.

Problem running '/etc/ufw/before6.rules'
```

I then ran 
`sudo check-requirements`
under ./usr/share/ufw/

and all tests report problems on the ipv6 part

anyone have any suggestions?

Note:
same isssue wuth release v01.02.02
if i disabled the ipv6, all is OK

jan-kiszka, 2022-10-18:
You likely need to configure some extra kernel features for IPv6. Does the report tell more about what is missing?

jan-kiszka, 2023-06-19:
Please answer my question and re-open if still relevant.

closed 2023-06-19 by jan-kiszka