siemens/meta-iot2050 #368: Ethernet Port X1P2 RX is 0

applea9, 2022-10-17:
Hi,

I met a problem with IOT2050's Ethernet Port. 

The network configuration is X1P1 port is in DHCP mode, X1P2 port is in static IP mode.  My PC is connected to X1P2 port, and in the same subnet with X1P2. After reboot several times, I found I can't ping the IP of IOT2050 X1P2 port. Then I logged in the IOT2050 through X1P1 port, and I saw that the RX of X1P2 port is 0.

Could you help check this problem? Thanks.

jan-kiszka, 2022-10-18:
Anything of your solution worth the share with the community? Or was it related to local specifics?

applea9, 2022-10-20:
In order to avoid the issues caused by /etc/network/interfaces, I switch to use NetworkManger to manage the LAN port.

But I can still meet such problem sometimes. And the problem is always occurd on eno2. 

I use ifconfig and ethtool to get some diagnosis information, below is the inforamtion

[image]

[image]

[image]

muneale, 2022-10-27:
Hi everyone,
we are facing the same issue described here.
We have done a bunch of experiments and these are the results:

* This error happens only when no cable has been plugged on port X1P2 (at least for us);
* This error happens even when both iface are set with static IPs;
* This error happens randomly after few reboots.

We are using NetworkManager service, hardware is IoT2050 PG2 and the OS is based v1.2.2.

Any ideas how to solve this?

AsuraZeng, 2022-11-01:
> 
hi 
can you share your configuration in the /etc/network/interface.d?

AsuraZeng, 2022-11-01:
> Hi everyone, we are facing the same issue described here. We have done a bunch of experiments and these are the results:
> 
>     * This error happens even when both iface are set with static IPs;
> 
>     * This error happens randomly after few reboots.
> 
> 
> We are using NetworkManager service, hardware is IoT2050 PG2 and the OS is based v1.2.2.
> 
> Any ideas how to solve this? If there is anything that could help you to solve this problem, just ask.

hi @muneale 

I want to know how do you configure your IP address?

muneale, 2022-11-01:
> hi @muneale
>
> I want to know how do you configure your IP address?

Hi @AsuraZeng !

So, when the device is booted/rebooted the NetworkManager service automatically loads each connection's configuration that is inside the folder `/etc/NetworkManager/system-connections`.

If a user needs to modify some properties (e.g. the IP address), it uses a custom service that communicates with NetworkManager through DBus API.

To be more precise:
1. The custom service retrieves the connection configuration by query the NetworkManager through DBus.
2. A configuration's copy is modified according to user input.
3. The service deactivates the current connection, applies the modified configuration and reactivates the connection (always through DBus API).

Every changes to the connection's parameters is also applied to its configuration file, so at every reboot the changes will persist over time.

Currently the X1P2 configuration's file is the following (I just replaced id and uuid values):
```
[connection]
id=<my-id>
uuid=<my-uuid>
type=ethernet
interface-name=eth1
permissions=

[ethernet]
mac-address-blacklist=

[ipv4]
address1=192.168.10.100/24
address2=192.168.253.252/24
dns-search=
method=manual

[ipv6]
addr-gen-mode=stable-privacy
dns-search=
method=ignore
```

Lastly, I'm sure that this bug is not related to my custom service since:
* It doesn't appears when the user modifies the connection's parameters without rebooting the device.
* When it doesn't shows up, the connection is loaded correctly and everything works fine.

AsuraZeng, 2022-11-02:
hi @muneale  
could you pls provide what's the image version you use?

use
```
cat /etc/os-release
```

muneale, 2022-11-02:
> hi @muneale could you pls provide what's the image version you use?
> 
> use
> 
> ```
> cat /etc/os-release
> ```

This is the output of the command that you provided:
```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
BUILD_ID="V01.02.01-15-gc54fe24"
VARIANT="IOT2050 Debian Example Image"
VARIANT_VERSION="1.0"
```

AsuraZeng, 2022-11-02:
> This is the output of the command that you provided:
>
> ```
> PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
> NAME="Debian GNU/Linux"
> VERSION_ID="11"
> VERSION="11 (bullseye)"
> VERSION_CODENAME=bullseye [...]

I have reproduced this problem. I would look into this issue.

muneale, 2022-11-02:
@AsuraZeng thank you so much! Let me know if you need anything else.

BTW, I also opened a topic on the Siemens support ([link here](https://support.industry.siemens.com/forum/WW/en/posts/port-x1p2-not-works-after-several-reboots/290172/?page=0&pageSize=10)).
They stated this:
```
[...] according to R&D this behavior is known and a bug we currently have.
R&D is working on it, but it is really hard as it is a sporadic issue and has no real pattern to reproduce.
```
And also:
```
R&D did the tests with only one interface set to dhcp.
So it seems that it is not only your connection settings, but also others. 
Besides the reboot nobody could tell me another workaround unfortunately. 
I hope they will find out something soon
```

muneale, 2022-12-12:
@AsuraZeng any news about the issue?

AsuraZeng, 2022-12-12:
> @AsuraZeng any news about the issue?

hi meueal

After catching up the icssg driver. it's ok.
But this is needing more tests. currently, we are planning to do the tests.

you can refer https://github.com/siemens/meta-iot2050/pull/374

applea9, 2023-02-15:
Is this issue solved by PR[https://github.com/siemens/meta-iot2050/pull/411](https://github.com/siemens/meta-iot2050/pull/411)?

jan-kiszka, 2023-02-15:
We are still seeing sporadic issues after reboots (view in hundreds of them). If you still have a way to reproduce your issue, it would still be worth revalidating with that update. It's in line right now with latest BSP patches from TI, but that may not be yet sufficient for all setups.

applea9, 2023-02-16:
ok, I will try the updated version. So does it mean that further improvement or bug fix for this issue will depend on future BSP patches form TI?

muneale, 2023-02-16:
I will also give it a try, thank you for your job.
Also, could you tell us approximately when this patch will be part of an official release?

SebaNar, 2025-02-04:
The problem occurs with current firmware and example image 1.4.4 again.
When both ports are configured static then eno2 half time of reboots not getting to work.

@jan-kiszka 
Please reopen the issue.

rFond, 2026-10-01:
We reproduced this issue on an IOT2050 Advanced PG1 (AM65x SR1.0), firmware V01.06.03, running kernel 6.12.90-cip24, and found a deterministic race in `icssg-prueth-sr1`.

**Symptom**

After a boot, either following a software reboot or a power cycle, one Ethernet port may remain unable to receive traffic for the entire boot.

The physical link is up at 1 Gb/s and the MAC RX counters increase, but no packet reaches the Linux networking stack. An AF_PACKET capture also sees no traffic, including in promiscuous mode.

In the failed state, the RX data IRQ counter remains at exactly one interrupt.

**Root cause**

In `emac_ndo_open()`, the default classifier is configured and the PRU/RTU firmware is started before the RX path is fully ready and before NAPI is enabled.

On SR1.0, the firmware can forward received frames to the host immediately after startup.

If traffic is already present on the Ethernet link, the first RX interrupt can therefore arrive while NAPI is still in the disabled state inherited from `netif_napi_add()`.

`prueth_rx_irq()` then disables the RX IRQ and calls `napi_schedule()`, but scheduling does not take place because NAPI is not enabled yet. Consequently the NAPI poll handler never runs and the IRQ is never re-enabled.

The interface then remains permanently unable to receive traffic until it is initialized again.

We reproduced the issue on SR1.0. We have not reproduced it on SR2; its initialization/forwarding sequence appears not to expose the same window.

**Reproduction**

With traffic present on the cable while the interface is opened:

- 4 failed boots out of 4.

Opening the same interface without a cable connected and plugging the cable in afterward works immediately.

**Fix**

We keep the classifier disabled while the RX path is initialized and only program the default classifier once the RX channels and NAPI are ready, mirroring the shutdown sequence.

Results with the fix:

- 14 successful boots out of 14
- 11 software reboots
- 3 power cycles
- traffic continuously present during interface initialization

We are preparing the corresponding patch for submission to the Linux networking maintainers and will link it here once submitted.

There is also a related shutdown issue: bringing an interface down while it is in this RX-stalled state can trigger:

`k3_knav_desc_pool size 65536 != avail 0`

followed by a kernel BUG in `lib/genalloc.c`.

This appears to be the same failure reported in #536. We have a separate fix for that path as well.

rFond, 2026-10-05:
Follow-up to my previous comment: here are both fixes, as we now ship them in our product image (IOT2050 Advanced PG1 and PG2, kernel 6.12.90-cip24 from meta-iot2050).

### 1. RX dead for the whole boot (SR1.0 open race)

`emac_ndo_open()` in `icssg_prueth_sr1.c` keeps the classifier **disabled** while the firmware starts and RX is set up, and only programs the default classifier **after** `napi_enable(&emac->napi_rx)`, so no frame from the wire can raise the first RX interrupt before NAPI is armed.

Bench results with the fix and traffic present on the port while it opens: 14 healthy boots out of 14 (11 software reboots, 3 power cycles), both ports receiving. Without it: 4 dead ports out of 4.

<details>
<summary>0001-net-ti-icssg-prueth-sr1-open-the-classifier-only-once-RX-is-ready.patch</summary>

```diff
From 0000000000000000000000000000000000000000 Mon Sep 17 00:00:00 2001
From: Clauger 3E <dev@clauger.fr>
Date: Thu, 1 Oct 2026 14:30:00 +0200
Subject: [PATCH] net: ti: icssg-prueth-sr1: open the classifier only once RX
 is ready

emac_ndo_open() programs the default classifier first and boots the
PRU/RTU firmware before the RX channels are enabled and before
napi_enable(). Unlike the SR2 firmware, which forwards nothing until
ICSSG_EMAC_PORT_FORWARD is set on link up, the SR1 firmware pushes
received frames to the host as soon as it runs. With traffic on the wire,
the first RX interrupt can therefore fire while NAPI_STATE_SCHED is still
set by netif_napi_add(): prueth_rx_irq() disables the IRQ, napi_schedule()
does nothing, and the poll that would re-enable the IRQ never runs. RX is
dead until the next open.

Observed on an IOT2050 Advanced (AM65x SR1.0): MAC statistics keep
counting received frames, the RX data IRQ count stays at exactly 1, and
nothing reaches the stack for the whole boot. Reproduced on 4 consecutive
boots with traffic on the port at open time; opening the same port
without link, then plugging the cable, works. With this patch: 14 out of
14 boots with traffic at open time, both ports receiving.

Keep the classifier disabled while the firmware starts and RX is set up,
and program the default classifier only once the RX channels and NAPI
are enabled, mirroring emac_ndo_stop() which disables it first to block
packets from the wire.

---
--- a/drivers/net/ethernet/ti/icssg/icssg_prueth_sr1.c
+++ b/drivers/net/ethernet/ti/icssg/icssg_prueth_sr1.c
@@ -500,7 +500,13 @@
 
 	icssg_class_set_mac_addr(prueth->miig_rt, slice, emac->mac_addr);
 
-	icssg_class_default(prueth->miig_rt, slice, 0, true);
+	/* The SR1 firmware pushes frames to the host as soon as it runs: an
+	 * RX interrupt raised before napi_enable() masks the IRQ for good
+	 * (NAPI_STATE_SCHED still set, the poll never runs). Keep frames from
+	 * the wire away until RX channels and NAPI are enabled; the classifier
+	 * is opened then, mirroring emac_ndo_stop().
+	 */
+	icssg_class_disable(prueth->miig_rt, slice);
 
 	/* Notify the stack of the actual queue counts. */
 	ret = netif_set_real_num_tx_queues(ndev, num_data_chn);
@@ -599,6 +605,8 @@
 		napi_enable(&emac->tx_chns[i].napi_tx);
 	napi_enable(&emac->napi_rx);
 
+	icssg_class_default(prueth->miig_rt, slice, 0, true);
+
 	/* start PHY */
 	phy_start(ndev->phydev);
 
```
</details>

[... 113 lines cut ...]
 	cancel_delayed_work_sync(&emac->stats_work);
 
 	/* stop PRUs */
 	prueth_emac_stop(emac);
 
-	free_irq(emac->rx_mgm_chn.irq[PRUETH_RX_MGM_FLOW_TIMESTAMP_SR1], emac);
-	free_irq(emac->rx_mgm_chn.irq[PRUETH_RX_MGM_FLOW_RESPONSE_SR1], emac);
 	free_irq(emac->rx_chns.irq[rx_flow], emac);
 	prueth_ndev_del_tx_napi(emac, emac->tx_ch_num);
 	prueth_cleanup_tx_chns(emac);
```
</details>

### Applying them in a meta-iot2050 based image

```bitbake
# recipes-kernel/linux/linux-iot2050_%.bbappend
FILESEXTRAPATHS:prepend := "${THISDIR}/files:"
SRC_URI += " \
    file://0001-net-ti-icssg-prueth-sr1-open-the-classifier-only-once-RX-is-ready.patch \
    file://0002-net-ti-icssg-prueth-reclaim-completed-RX-descriptors-on-stop.patch \
"
```

@jan-kiszka since this issue is closed but the defect is still present on current releases, would you consider reopening it, or taking these patches into the BSP?

BaochengSu, 2026-10-08:
Thanks for your reporting and fixing of the issue. I will file an PR for these patches into this repo.

PR #411 (2023-02-15): Update TI bits in linux-iot2050
Align the BSP kernel regarding its TI downstream or backport bits with latest ti-linux. Also update the prueth firmware along that. Should resolve #368 and supersedes #374.

This does not including a stable update of the underlying CIP kernel yet as we are waiting for a recent cip-rt release.