siemens/meta-iot2050 #384: Modbus FS05

Codfired, 2022-11-23:
Hello everbody,

i have the same problem with RS485 on X30 Connector. Where can i become the new image without building ?

My Setup:

IOT2050 Advance FS: 01 with "IOT2050_Example_Image_V1.2.2.zip (780,6 MB)" from Entry-ID: 109741799
-> RS485 work
-> iot2050setup availiable

IOT2050 Advance FS: 05 with "IOT2050_Example_Image_V1.2.2.zip (780,6 MB)" from Entry-ID: 109741799
-> RS485 not work

-> iot2050setup not availiable

tested with: pymodbus and mbpoll

transfer Modbus-Config over "switchserialmode cp210x -D CP2102N24 -r/w seriel_config.txt" from FS01 to FS05
-> Modbus not work

Have issue with Image ? or other Pinout between FS01 and FS05 ? (1: Data+,2: Data-)

[image]

jan-kiszka, 2022-11-23:
If you are lucky, you will find artifacts from our build tests under Actions, see e.g. https://github.com/siemens/meta-iot2050/actions/runs/3510848329.

And if you do not have iot2050setup in your image, you have the Industrial OS in hands which is not distributed or built here. What does /etc/os-release report on your images?

Codfired, 2022-11-24:
Now /etc/os-release reports:

[image]

jan-kiszka, 2022-11-24:
That's clearly this example image, release state. It contains an iot2050setup tool.

jan-kiszka, 2022-11-24:
BTW, there has just been a fix for the setup/host time merged, see https://github.com/siemens/meta-iot2050/pull/375

closed 2022-11-29 by Codfired