siemens/meta-iot2050 #385: version of the iot2050-setup tool never bumped

chombourger, 2022-11-24:
The tool was introduced 3 years ago with version 0.1
several features were added since then but the version is still 0.1 and several versions of this distribution were released
versions (and changelogs) should be maintained especially for binary distributions that are consuming your recipes (e.g. to support apt dist-upgrade scenario)

jan-kiszka, 2022-11-24:
We likely have that issue not only in this context, possibly also for other self-written packages. Any suggestions how to make sure that we do not miss recipe version bumps on tool updates?

For this tool, it would probably the first step to add a version to it that is also somewhere tracked in the code - and then reflected also in the recipe.

jan-kiszka, 2023-10-23:
iot2050-setup became board-conf-tools and that did not run into the same issue so far.

closed 2023-10-23 by jan-kiszka