siemens/meta-iot2050 #386: node-red: ./src/threadpool.c:329: uv__queue_done: Assertion `uv__has_active_reqs(req->loop)' failed

bergmanu, 2022-11-28:
A customer in the [forum ](https://support.industry.siemens.com/forum/ww/en/posts/assertion-uv-has-active-reqs-req-loop-failed/289029/?page=0&pageSize=10) reported that his node-red application crashes periodically when he reads four inputs in a 7.5Hz period. 
The crash reason is:
`node-red: ./src/threadpool.c:329: uv__queue_done: Assertion `uv__has_active_reqs(req->loop)' failed`

I reproduced that issue with a simplified flow (see attachment): Only having three digital inputs connected
[DI_crash_simplified.txt](https://github.com/siemens/meta-iot2050/files/10102366/DI_crash_simplified.txt)
 and provide a "blink" signal in a 7.5Hz period by DQs of a S7-1500 PLC. 
 (Mostly) As soon as the blink application starts, node-red crashes with the following statement: 

```

Jul 13 18:19:50 iot2050-debian node-red[818]: node-red: ./src/threadpool.c:329: uv__queue_done: Assertion `uv__has_active_reqs(req->loop)' failed.
Jul 13 18:19:50 iot2050-debian systemd[1]: node-red.service: Main process exited, code=killed, status=6/ABRT
Jul 13 18:19:50 iot2050-debian systemd[1]: node-red.service: Failed with result 'signal'.
Jul 13 18:19:50 iot2050-debian systemd[1]: node-red.service: Consumed 29.504s CPU time.

```

Changing the period to 1Hz does not change the behavior.

Hardware / Software used: 

- IOT2050 Basic (FS:02)
- Input Module Sink/Source
- Example Image V1.2.2

jan-kiszka, 2022-11-28:
Reproduced, also with the board generating the signals itself (I don't have a PLC around). Seems there is some race condition, hopefully in mraa (which is easier to fix) and not in libuv, the nodejs event helper library. Needs more debugging.

BTW, it seems the issue was also reported to Node-RED before, see https://discourse.nodered.org/t/assertion-uv-has-active-reqs-req-loop-failed/68564.

jan-kiszka, 2022-12-04:
See https://github.com/libuv/libuv/issues/3846 for a first attempt to fix it. Looks good here with 50 Hz self-trigger on two inputs which crashes immediately otherwise.

jan-kiszka, 2022-12-05:
A (likely) more correct fix is under testing now, see https://github.com/siemens/mraa/commit/03af3e87e8f43cbd4450a59d829b5468e6b53f9b.

bergmanu, 2022-12-06:
Can confirm that the image containing this fix does not have this behavior anymore. 
Tested the flow with 3 connected DIs (7,5Hz) without any error for 2h.

commit 35c4a589 (2022-12-12): mraa: Add patch to fix "uv__has_active_reqs" assertions

See the included patch for more details. Related mraa PR is
https://github.com/eclipse/mraa/pull/1102.

Fixes #386
closed 2022-12-12 by BaochengSu