siemens/meta-iot2050 #407: Different AI resolution in python and node-red

bergmanu, 2023-01-12:
Using the GPIO node-red node to read in an analog value from the I/O Module - either 0..10V or 0..20mA - results in a value from 0..4095 (12-bit resolution)

Using the same analog input with mraa in python the resolution is only 0..1023 (9-bit)

Where does this difference come from?

AsuraZeng, 2023-01-12:
After my investigation, node red set the ADC bit to 12bit, but
the default setting in mraa is 10bit. The current workaround is to set the bit of adc to 12 after initializing the ADC.

I have a PR to resolve this one.
https://github.com/eclipse/mraa/pull/1103.
it's under review currently.

jan-kiszka, 2023-01-13:
As you correctly noted: That PR is a workaround, not the solution yet.

jan-kiszka, 2023-01-13:
And: there is no issue. node-red-node simply raises the number of bits to what the hardware can do. That is done at https://github.com/node-red/node-red-nodes/blob/9bae313b722147b00480ce2e26ed0aab357b836b/hardware/intel/mraa-gpio-ain.js#L15. That is logically one level above what the Python (or also NodeJS) bindings does which only exposes the AIO abstraction. And that abstraction chose to start with 10 bits, hardware-independently.

IOW: Nothing to fix here or elsewhere, just to understand logic-wise.

closed 2023-01-13 by jan-kiszka

jan-kiszka, 2023-01-13:
In that light, setting the number of bits unconditionally in node-red-node could even be questioned from a hardware abstraction POV. It should rather become a tunable of that node.