siemens/meta-iot2050 #418: iot2050-image-example.bb:do_rootfs_install failed

lxmuyu, 2023-02-16:
when we build the image,we get the error:'Packages were downgraded and -y was used without --allow-downgrades',and it has some info 'The following packages will be DOWNGRADED:
  libssl1.1
0 upgraded, 571 newly installed, 1 downgraded, 0 to remove and 0 not upgraded.
E: Packages were downgraded and -y was used without --allow-downgrades.
WARNING: exit code 100 from a shell command.' . we clear the project and rebuild, the error is same.

How can we fix it ? 

branch:origin/stable/V01.02 
build cmd: ./kas-container build kas-iot2050-example.yml:kas/opt/mirror-example.yml
error msg:  Packages were downgraded and -y was used without --allow-downgrades
log:
`
2023-02-16 02:22:22 - INFO     - kas 2.5 started
2023-02-16 02:22:22 - INFO     - /repo$ git rev-parse --show-toplevel
2023-02-16 02:22:22 - INFO     - /repo/kas/opt$ git rev-parse --show-toplevel
2023-02-16 02:22:22 - INFO     - /repo$ git rev-parse --show-toplevel
2023-02-16 02:22:22 - INFO     - Using /repo as root for repository meta-iot2050
2023-02-16 02:22:22 - INFO     - /work/isar$ git remote set-url origin https://github.com/ilbers/isar
2023-02-16 02:22:22 - INFO     - /work/isar$ git cat-file -t e274130b870b7e31532fe191b9932cde1d819b4b
2023-02-16 02:22:22 - INFO     - Repository isar already contains e274130b870b7e31532fe191b9932cde1d819b4b as commit
2023-02-16 02:22:22 - INFO     - /repo$ git rev-parse --show-toplevel
2023-02-16 02:22:22 - INFO     - Using /repo as root for repository meta-iot2050
2023-02-16 02:22:22 - INFO     - /work/isar$ git status -s
2023-02-16 02:22:22 - INFO     - /work/isar$ git rev-parse --verify -q origin/e274130b870b7e31532fe191b9932cde1d819b4b
2023-02-16 02:22:22 - INFO     - /work/isar$ git checkout -q e274130b870b7e31532fe191b9932cde1d819b4b
2023-02-16 02:22:22 - INFO     - /repo$ git rev-parse --show-toplevel
2023-02-16 02:22:22 - INFO     - Using /repo as root for repository meta-iot2050
2023-02-16 02:22:22 - INFO     - /work/isar$ git checkout -q -B patched-e274130b870b7e31532fe191b9932cde1d819b4b
2023-02-16 02:22:22 - INFO     - /work/isar$ git apply /repo/isar-patches/0001-u-boot-Exclude-libubootenv.a-from-dh_strip.patch
2023-02-16 02:22:22 - INFO     - Patch applied. (patch path: /repo/isar-patches/0001-u-boot-Exclude-libubootenv.a-from-dh_strip.patch, repo: isar, patch entry: u-boot-fix)
2023-02-16 02:22:22 - INFO     - /work/isar$ git add -A
2023-02-16 02:22:22 - INFO     - /work/isar$ git commit -a --author kas <kas@example.com> -m msg
2023-02-16 02:22:22 - INFO     - [patched-e274130b870b7e31532fe191b9932cde1d819b4b eee20ab] msg
2023-02-16 02:22:22 - INFO     - Author: kas <kas@example.com>
2023-02-16 02:22:22 - INFO     - 1 file changed, 3 insertions(+)
2023-02-16 02:22:22 - INFO     - /repo$ git rev-parse --show-toplevel
2023-02-16 02:22:22 - INFO     - Using /repo as root for repository meta-iot2050
2023-02-16 02:22:22 - INFO     - /work/isar$ /tmp/tmpf2xiqzjm/get_bb_env /build
2023-02-16 02:22:22 - INFO     - /repo$ git rev-parse --show-toplevel
2023-02-16 02:22:22 - INFO     - Using /repo as root for repository meta-iot2050
2023-02-16 02:22:22 - INFO     - /build$ /work/isar/bitbake/bin/bitbake -c build iot2050-image-example
Loading cache: 100% |###########################################################################################################################################################################################################################| Time: 0:00:00
Loaded 53 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
Initialising tasks: 100% |######################################################################################################################################################################################################################| Time: 0:00:00
NOTE: No setscene tasks
NOTE: Executing Tasks
ERROR: iot2050-image-example-1.0-r0 do_rootfs_install: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_rootfs_install(d)
     0003:
File: '/work/isar/meta/classes/rootfs.bbclass', lineno: 182, function: do_rootfs_install
     0178:                                     shared=True)
     0179:
     0180:        bb.build.exec_func(cmd, d)
     0181:
 *** 0182:        if (d.getVarFlag(cmd, 'isar-apt-lock') or "") == "release-after":
     0183:            bb.utils.unlockfile(lock)
     0184:    progress_reporter.finish()
     0185:}
     0186:addtask rootfs_install before do_rootfs_postprocess after do_unpack
File: '/work/isar/bitbake/lib/bb/build.py', lineno: 251, function: exec_func
     0247:    with bb.utils.fileslocked(lockfiles):
     0248:        if ispython:
     0249:            exec_func_python(func, d, runfile, cwd=adir)
     0250:        else:
[... 259 lines cut ...]
  openjdk-11-jre-headless openssh-client openssh-server openssh-sftp-server
  openssl patch pci.ids pciutils perl perl-modules-5.32 policykit-1 ppp procps
  psmisc pulseaudio pulseaudio-module-bluetooth pulseaudio-utils
  python-is-python3 python3 python3-minimal python3-newt python3-pkg-resources
  python3.9 python3.9-minimal regen-rootfs-uuid resolvconf rsyslog rt-tests
  runit-helper sensible-utils shared-mime-info ssh sshd-regen-keys stress-ng
  sudo switchserialmode systemd systemd-sysv tcf-agent ti-pruss-firmware
  u-boot-iot2050-config u-boot-script u-boot-tools ucf udev usbutils
  uuid-runtime vim vim-common vim-runtime wget wpasupplicant x11-common
  xdg-utils xxd xz-utils
The following packages will be DOWNGRADED:
  libssl1.1
0 upgraded, 571 newly installed, 1 downgraded, 0 to remove and 0 not upgraded.
E: Packages were downgraded and -y was used without --allow-downgrades.
WARNING: exit code 100 from a shell command.

ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/iot2050-image-example-iot2050-wic-img/1.0-r0/temp/log.do_rootfs_install.77
ERROR: Task (/repo/recipes-core/images/iot2050-image-example.bb:do_rootfs_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 409 tasks of which 408 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /repo/recipes-core/images/iot2050-image-example.bb:do_rootfs_install
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
2023-02-16 02:22:33 - ERROR    - Command returned non-zero exit status 1
`

BaochengSu, 2023-02-16:
Hi, 

If you are building the stable branch, better to have the `kas/opt/package-lock.yml` in the build command.

See https://github.com/siemens/meta-iot2050/tree/stable/V01.02#build-released-version

lxmuyu, 2023-02-16:
> Hi,
> 
> If you are building the stable branch, better to have the `kas/opt/package-lock.yml` in the build command.
> 
> See https://github.com/siemens/meta-iot2050/tree/stable/V01.02#build-released-version

Thank you for your reply, I'll try it now. There are results for you to reply.

lxmuyu, 2023-02-17:
@BaochengSu After a long build process, we can generate images, thank you! However, installing apts from `snapshot.debian.org` is unstable, and the package installation timeout is serious ...This is another question. Thanks!

closed 2023-02-17 by lxmuyu

BaochengSu, 2023-02-17:
Glad you've done that.

The reproducible of the stable version image has been an outstanding issue for pretty a while, currently s.d.o throttles the bandwidth and the connections to avoid abuse.

However, some community efforts have been already paid to address such an issue, e.g. https://debconf21.debconf.org/talks/22-making-use-of-snapshotdebianorg-for-fun-and-profit/. 

Some other workaround I found useful is to use some apt cacher (such as apt-cacher-ng) to cache the .deb packages, although it does not solve the first time downloading, it makes the following buildings more smoothly. However you need time and computing resource to implement such cache proxy.