siemens/meta-iot2050 #422: Iot2050 factory reset

Mrkky, 2023-03-08:
Hi,
How can I do reset the iot2050 to factory settings? (eMMC)

BaochengSu, 2023-03-09:
Hi,

What do you want to achieve? to restore the factory Industrial OS image? If the eMMC is already overwritten by other os image, I think the only way is to re-install Industrial OS.

Mrkky, 2023-03-09:
Hi,

My main problem is I can't connect to iot2050 device by IP. I'm just connecting from the serial port with the uart cable. I have been trying to solve this problem for months, Siemens has not found a solution yet. I guess if I reset emmc and reinstall it maybe it will fix.

BaochengSu, 2023-03-09:
What is the OS now? Industrial OS or example image, or customized OS? It could be determined by checking `/etc/os-release`?

By the way, if that is the Industrial OS pre-installed, I guess you need a DP cable and a monitor to do the initial setup, for example the IP and the user name.

There is a default static IP 192.168.200.1 in example image. I am not familiar with the Industrial OS, so I am not sure if there is any such default IP address for Industrial OS, but I guess if you've configured the device, there will be some IP available.

Also make sure the service is running on the IOT2050 to be connected to, such as SSH or web server.

bergmanu, 2023-03-09:
Can you check nmcli when being connected via UART as well?

Mrkky, 2023-03-09:
root@localhost:~# nmcli
eth0: connected to eth0-connection
        "eth0"
        ethernet (icssg-prueth), 8C:F3:19:02:57:41, hw, mtu 1500
        inet4 192.168.200.1/24
        route4 192.168.200.0/24
        inet6 fe80::80ac:46c8:6e8:c8b/64
        route6 fe80::/64
        route6 ff00::/8

eth1: unavailable
        "eth1"
        ethernet (icssg-prueth), 8C:F3:19:02:57:42, hw, mtu 1500

lo: unmanaged
        "lo"
        loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536

Use "nmcli device show" to get complete information about known devices and
"nmcli connection show" to get an overview on active connection profiles.

Consult nmcli(1) and nmcli-examples(5) manual pages for complete usage details.
root@localhost:~#

bergmanu, 2023-03-09:
I now got the connection to the corresponding [Forum post](https://support.industry.siemens.com/forum/ww/en/posts/iot2050-connection-problem/295711/?page=2&pageSize=10#post1136026)
I answered in the forum right now. Please try this

Mrkky, 2023-03-09:
Hi,
The problem is solved.
thank you.

closed 2023-03-09 by Mrkky