siemens/meta-iot2050 #423: Filled SMBIOS part in DTB file

sgthammer, 2023-03-13:
### As UEFI for arm64 is available by u-boot, for the purpose of unified platform retrieval for different architectures, DMI tables should be available for IOT2050/ARM64 device as in the x86_64 devices.

- _DMI types are available under this link:_ [DMI types](https://linux.die.net/man/8/dmidecode)
If it is possible, please add as many as of them.
- Therefore, please fill smbios part of DTB file for IOT2050 device ([smbios.txt](https://u-boot.readthedocs.io/en/latest/develop/smbios.html)).

Thanks in advance.

jan-kiszka, 2023-06-09:
Sorry for the delay for this "simple" thing, we are on it.

A first upstream-aiming patch set can be found at https://github.com/siemens/u-boot/commits/jan/iot2050. An open question is which fields we must have in addition to this:
```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.0 present.
7 structures occupying 319 bytes.
Table at 0xFCED0020.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
        Vendor: U-Boot
        Version: 2023.07-rc3-00038-g6409bcf6933-dirty
        Release Date: 07/01/2023
        ROM Size: 64 kB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                Selectable boot is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 23.7

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: Siemens
        Product Name: SIMATIC IOT2050 Advanced PG2
        Version: Not Specified
        Serial Number: Not Specified
        UUID: Not Settable
        Wake-up Type: Reserved
        SKU Number: Not Specified
        Family: SIMATIC IOT2050

Handle 0x0002, DMI type 2, 14 bytes
Base Board Information
        Manufacturer: Siemens
        Product Name: SIMATIC IOT2050 Advanced PG2
        Version: Not Specified
        Serial Number: Not Specified
        Asset Tag: Not Specified
        Features:
                Board is a hosting board
        Location In Chassis: Not Specified
        Chassis Handle: 0x0000
        Type: Motherboard

Handle 0x0003, DMI type 3, 21 bytes
Chassis Information
        Manufacturer: Siemens
        Type: Desktop
        Lock: Not Present
        Version: Not Specified
        Serial Number: Not Specified
        Asset Tag: Not Specified
        Boot-up State: Safe
        Power Supply State: Safe
        Thermal State: Safe
        Security Status: None
        OEM Information: 0x00000000
        Height: Unspecified
        Number Of Power Cords: Unspecified
        Contained Elements: 0

Handle 0x0004, DMI type 4, 48 bytes
Processor Information
        Socket Designation: Not Specified
        Type: Central Processor
        Family: Unknown
        Manufacturer: Unknown
        ID: 00 00 00 00 00 00 00 00
        Version: Unknown
        Voltage: Unknown
        External Clock: Unknown
        Max Speed: Unknown
        Current Speed: Unknown
        Status: Unpopulated
        Upgrade: None
        L1 Cache Handle: Not Provided
        L2 Cache Handle: Not Provided
        L3 Cache Handle: Not Provided
        Serial Number: Not Specified
        Asset Tag: Not Specified
        Part Number: Not Specified
        Characteristics: None

Handle 0x0005, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected

Handle 0x0006, DMI type 127, 4 bytes
End Of Table
```
@BaochengSu

BaochengSu, 2023-06-12:
@jan-kiszka For the fields, I think we have to refer to the standard x86 IPCs and try to use the same field when possible.

And most fields are sourcing from the scratchpad iot2050-info, so I am working a sysinfo driver for iot2050 so that this information could be easily passed into the dmitable. Will send out the patch for reviewing and upstreaming once done, (now blocked by a sync data abort exception issue ...)

jan-kiszka, 2023-06-12:
Will that driver cover static elements from a DT input like above as well?

In any case, the driver alone does not answer what we fill on, only how.

BaochengSu, 2023-06-12:
It will, the sysinfo uclass in u-boot allows you to add the static elements in the DT, together with the dynamic elements from somewhere such as the eeprom to feed to the dmitable. I've tried that in my dirty branch, it worked.

see [smbios.txt](https://u-boot.readthedocs.io/en/latest/develop/smbios.html), esp. the example.

jan-kiszka, 2024-08-15:
FWIW, we finally have some driver and integration that should address this, see https://github.com/siemens/u-boot/commits/jan/iot2050. It's a preview and also still needs to be run via upstream, but feedback on the chosen (or missing) fields is welcome. Here some example:

```
root@iot2050-debian:~# dmidecode 
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.7.0 present.
# SMBIOS implementations newer than version 3.5.0 are not
# fully supported by this version of dmidecode.
Table at 0xFDF0E020.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: U-Boot
        Version: 2024.10-rc2-00138-gdd5a8638ce15
        Release Date: 10/01/2024
        ROM Size: 64 kB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                Selectable boot is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 24.10

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: SIEMENS AG
        Product Name: SIMATIC IOT2050
        Version: 9MC1110-7UG00-0AA0
        Serial Number: R3S75099
        UUID: 1c557e38-cdf3-4b4d-a4c9-1d39377c6a00
        Wake-up Type: Unknown
        SKU Number: Not Specified
        Family: Not Specified

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: SIEMENS AG
        Product Name: A5E520802600AB02
        Version: IOT2050-ADVANCED-SM
        Serial Number: Not Specified
        Asset Tag: Not Specified
        Features:
                Board is a hosting board
        Location In Chassis: Not Specified
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0

Handle 0x0003, DMI type 3, 21 bytes
Chassis Information
        Manufacturer: Not Specified
        Type: Desktop
        Lock: Not Present
        Version: Not Specified
        Serial Number: Not Specified
        Asset Tag: Not Specified
        Boot-up State: Safe
        Power Supply State: Safe
        Thermal State: Safe
        Security Status: None
        OEM Information: 0x00000000
        Height: Unspecified
        Number Of Power Cords: Unspecified
        Contained Elements: 0

Handle 0x0004, DMI type 4, 48 bytes
Processor Information
        Socket Designation: Not Specified
        Type: Central Processor
        Family: Unknown
        Manufacturer: Not Specified
        ID: 00 00 00 00 00 00 00 00
        Version: Not Specified
        Voltage: Unknown
        External Clock: Unknown
        Max Speed: Unknown
        Current Speed: Unknown
        Status: Unpopulated
        Upgrade: None
        L1 Cache Handle: Not Provided
        L2 Cache Handle: Not Provided
        L3 Cache Handle: Not Provided
        Serial Number: Not Specified
        Asset Tag: Not Specified
        Part Number: Not Specified
        Characteristics: None

Handle 0x0005, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected

Handle 0x0006, DMI type 127, 4 bytes
End Of Table

```
@huaqianli

colpane, 2024-11-25:
If it is possible, I would say to fill "Processor Information" part more. For example, it would be nice to get CPU Max Speed from this. @jan-kiszka @huaqianli