siemens/meta-iot2050 #425: Integrate .deb package

enzo-digex, 2023-03-14:
I know that it is an awful idea...but there is no other way:
We need to integrate a debian package into our yocto project. Can anyone give me a hint how such a recipe should look like? 

My bb file is: 

```
DESCRIPTION = "Installation deb package"

SRC_URI = "\ 
            file://mypackage.deb;unpack=0 \
          "

DEBIAN_DEPENDS += " dpkg-native "

do_install_append() {
    touch ${STAGING_DIR_NATIVE}/var/lib/dpkg/status
    ${STAGING_BINDIR_NATIVE}/usr/bin/dpkg --instdir=${D}/ \
    --admindir=${STAGING_DIR_NATIVE}/var/lib/dpkg/ \
     -i ${WORKDIR}/mypackage.deb
}
```
But this throws an error: 

```
Reading state information...
E: Unable to locate package mypackage
WARNING: exit code 100 from a shell command.
ERROR: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_rootfs_install(d)
     0003:
File: '/work/isar/meta/classes/rootfs.bbclass', lineno: 187, function: do_rootfs_install
     0183:                                     shared=True)
     0184:
     0185:        bb.build.exec_func(cmd, d)
     0186:
 *** 0187:        if (d.getVarFlag(cmd, 'isar-apt-lock') or "") == "release-after":
     0188:            bb.utils.unlockfile(lock)
     0189:    progress_reporter.finish()
     0190:}
     0191:addtask rootfs_install before do_rootfs_postprocess after do_unpack
File: '/work/isar/bitbake/lib/bb/build.py', lineno: 251, function: exec_func
     0247:    with bb.utils.fileslocked(lockfiles):
     0248:        if ispython:
     0249:            exec_func_python(func, d, runfile, cwd=adir)
     0250:        else:
 *** 0251:            exec_func_shell(func, d, runfile, cwd=adir)
     0252:
     0253:    try:
     0254:        curcwd = os.getcwd()
     0255:    except:
File: '/work/isar/bitbake/lib/bb/build.py', lineno: 452, function: exec_func_shell
     0448:    with open(fifopath, 'r+b', buffering=0) as fifo:
     0449:        try:
     0450:            bb.debug(2, "Executing shell function %s" % func)
     0451:            with open(os.devnull, 'r+') as stdin, logfile:
 *** 0452:                bb.process.run(cmd, shell=False, stdin=stdin, log=logfile, extrafiles=[(fifo,readfifo)])
     0453:        finally:
     0454:            os.unlink(fifopath)
     0455:
     0456:    bb.debug(2, "Shell function %s finished" % func)
File: '/work/isar/bitbake/lib/bb/process.py', lineno: 182, function: run
     0178:        if not stderr is None:
     0179:            stderr = stderr.decode("utf-8")
     0180:
     0181:    if pipe.returncode != 0:
 *** 0182:        raise ExecutionError(cmd, pipe.returncode, stdout, stderr)
     0183:    return stdout, stderr
Exception: bb.process.ExecutionError: Execution of '/build/tmp/work/iot2050-debian-arm64/iot2050-image-example-iot2050-wic-img/1.0-r0/temp/run.rootfs_install_pkgs_download.91' failed with exit code 100:
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package mypackage
```
I triple checked the situation...and the deb file is definitely in my recipe, but is not copied. So I suspect that my recipe is crap.

jan-kiszka, 2023-03-14:
Err, this here is not meta-iot2000, thus not a Yocto project - hope that is clear :wink:.

I think you want to study https://github.com/ilbers/isar/blob/master/meta-isar/recipes-app/prebuilt-deb/prebuilt-deb_0.1.bb for your problem. There is also https://github.com/ilbers/isar/blob/master/doc/user_manual.md#prebuilt-deb-packages-from-somewhere.

enzo-digex, 2023-03-14:
My apologies for this mistake...of course I that this is not a Yocto project ;-) 
Applying the prebuilt-deb example, it integrates the package. But at the moment it is preparing to unpack the package it comes to the error "Unsupported Linux distribution".
Well...I unpacked the *.deb file and looked in the DEBIAN/preinst file:
```
#!/bin/sh

# check OS type
if [ -d /run/systemd/system ]; then
    INIT_SYSTEM=systemd
else
    echo "Unsupported Linux distribution"
    exit 7
fi

# detect hardware platform
ARCH=`uname -m`
case "$ARCH" in
    armv7l)
        ;;
    aarch64)
        ;;
    *)
        echo "Unsupported hardware platform"
        exit 8
        ;;
esac

# in case of update scenario, otherwise it's an install
if [ "$1" = "2" -o "$1" = "upgrade" ]
then
    case $INIT_SYSTEM in
    systemd)
        systemctl stop hasplmd
        systemctl disable hasplmd
        ;;
    esac
fi
```
Obviously, it doesn't find the directory "/run/systemd/system"...How to get around this problem? Do I need to "patch" the deb package or is there any other possibility? I did not find any hint in the isar documentation...

jan-kiszka, 2023-03-15:
Yes, you will need to massage the binary package before the do_dpkg_build task. We don't have a published example for that yet (AFAIK), but it could look similar to this:
```
inherit dpkg-prebuilt

...

do_repackage[cleandirs] = "${WORKDIR}/repack-tmp"
do_repackage() {
    cd ${WORKDIR}/repack-tmp
    rm -rf *
    ar xo $MY_DEB_FILE

    mkdir control
    tar -C control -xf control.tar
    mkdir data
    tar -C data -xf data.tar.zst

    <massage control or data content>

    tar -C control --owner=0 --group=0 -cf control.tar .
    tar -C data --owner=0 --group=0 -cf data.tar .
    xz -T0 data.tar
    ar rc $MY_DEB_FILE debian-binary control.tar data.tar.xz
}
addtask repackage after do_unpack before do_dpkg_build
```

jan-kiszka, 2023-03-15:
BTW, that arch check of the preinst script is pretty pointless. You normally encode the supported arch into the meta data of the deb file. But, yeah... I've seen a lot of nonsense in downstream binary packages already.

enzo-digex, 2023-03-15:
> 

Yes...I know...but this is what we got from our supplier and customer wants this in his image...

Nevertheless...the do_repackage section was the missing part! thx for you help!

I close this issue

closed 2023-03-15 by enzo-digex