siemens/meta-iot2050 #427: Kernel OOPS while using rmmod for ethernet driver

sgthammer, 2023-03-28:
Hello,

For an image with swupdate support, I tried to remove ethernet driver module using _rmmod_. "**rmmod iccsg_prueth**" command is used for this aim and the _rmmod_ throws segmentation fault. Also, kernel throws OOPS as internal error. Kernel log file is attached.

What could be the reason causes this issue?
[oops.log](https://github.com/siemens/meta-iot2050/files/11090619/oops.log)

nhahnetsrot, 2023-05-24:
Pls. try the attached patch, needs to be copied to meta-iot2050/recipes-kernel/linux/files/patches-5.10/ and then the image needs to be rebuild (e.g.: sudo kas-container clean; kas-container build )

[0211-fix-rmmod.patch](https://github.com/siemens/meta-iot2050/files/11551349/0211-fix-rmmod.patch)

jan-kiszka, 2023-10-13:
Will be fixed by #491

closed 2023-10-23 by jan-kiszka