siemens/meta-iot2050 #432: Use dedicated base-schroot for NPM package builds

fmoessbauer, 2023-04-10:
Since the introduction of sbuild in ISAR, all `do_dpkg_build` tasks run in their on isolated environment. The base rootfs of the builds is shared, but the build itself uses an overlay so that multiple builds can work isolated and concurrently.

For NPM modules, a lot of packages need to be installed prior to the build. These packages are needed for all NPM builds, but are only installed into the overlay of the schroot. By that, a ton of logic is executed in each build (which also needs to be emulated due to missing cross support). This is very inefficient and disk-bound.

Instead, it would be better to create a dedicated schroot with all base packages for the NPM builds. Then, the build itself only needs to install the package specific dependencies and runs much quicker.

As this is becoming a pattern, we should also consider a small ISAR patch to make it easier to create derived schroots from either the host or the target base schroot.

jan-kiszka, 2023-04-12:
Sure - already have patches?

fmoessbauer, 2023-04-12:
> Sure - already have patches?

Not yet. I just had a brief look at the `sbuild-chroot.inc` file and it looks quite straight forward to implement. I can give it a try and send it to the ISAR ML.

PR #433 (2023-09-04): feat: use dedicated schroot to speedup npm builds
Let's wait till this has been integrated upstream.

closes: #432
closed 2023-09-04 by BaochengSu