siemens/meta-iot2050 #439: dpkg-shlibdeps: error: cannot find library libz.so.1

SCordibella, 2023-05-09:
Hi all, I am trying to package .NET runtime using dpkg-raw for arm64.
The Microsoft runtime is available for arm64, and I create a recipe adding the dependencies in DEBIAN_DEPENDS variable.
Here is my recipe:

```
DESCRIPTION = "ASPNET core 6.0.16 runtime"
MAINTAINER = "Stefano Cordibella <stefano.cordibella@edalab.it>"
DEBIAN_DEPENDS = "libc6, libgcc-s1, libgssapi-krb5-2, libicu67, liblttng-ust0, libssl1.1, libstdc++6, zlib1g"

SRC_URI = "https://download.visualstudio.microsoft.com/download/pr/5fe35f73-59e4-462e-b7aa-98b5b8782051/74a27e03d896663a9483eb72bc59b275/aspnetcore-runtime-6.0.16-linux-arm64.tar.gz"
SRC_URI[sha256sum] = "eba3d57245dcc02ef7b9677a6f205d8a0b932a4aa1e552c9e9094cefab6b52c2"

inherit dpkg-raw

PV = "6.0.16"

do_install() {
    DOTNET_ROOT=/usr/share/dotnet

    install -v -d ${D}/${DOTNET_ROOT}
    install -v -m 755 ${WORKDIR}/dotnet ${D}/${DOTNET_ROOT}
    install -v -m 744 ${WORKDIR}/LICENSE.txt ${D}/${DOTNET_ROOT}
    install -v -m 744 ${WORKDIR}/ThirdPartyNotices.txt ${D}/${DOTNET_ROOT}

    install -v -d ${D}/${DOTNET_ROOT}/shared
    cp -R --no-dereference --preserve=mode,links -v ${WORKDIR}/shared ${D}/${DOTNET_ROOT}
}
```

However when I build the package the build fails with the error:
```
dpkg-shlibdeps: error: cannot find library libz.so.1 needed by debian/dotnet-runtime/usr/share/dotnet/shared/Microsoft.NETCore.App/6.0.16/libSystem.IO.Compression.Native.so (ELF format: 'elf64-littleaarch64' abi: '020100b700000000'; RPATH: '')
```

Here is the complete log [dotnet.log](https://github.com/siemens/meta-iot2050/files/11428161/dotnet.log)

Looking in the build directory I find the libz.so.1 many times:
```
./tmp/work/iot2050-debian-arm64/sbuild-chroot-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/libz.so.1
./tmp/work/iot2050-debian-arm64/isar-bootstrap-host/1.0-r0/rootfs/usr/lib/x86_64-linux-gnu/libz.so.1
./tmp/work/iot2050-debian-arm64/iot2050-image-ckm-iot2050/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/libz.so.1
./tmp/work/iot2050-debian-arm64/iot2050-image-swu-example-iot2050/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/libz.so.1
./tmp/work/iot2050-debian-arm64/iot2050-initramfs-iot2050/0.1-r0/rootfs/usr/lib/aarch64-linux-gnu/libz.so.1
./tmp/work/iot2050-debian-arm64/isar-bootstrap-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/libz.so.1
./tmp/work/iot2050-debian-arm64/buildchroot-host/1.0-r0/rootfs/usr/lib/x86_64-linux-gnu/libz.so.1
./tmp/work/iot2050-debian-arm64/buildchroot-host/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/libz.so.1
./tmp/work/iot2050-debian-arm64/sbuild-chroot-host/1.0-r0/rootfs/usr/lib/x86_64-linux-gnu/libz.so.1
./tmp/work/iot2050-debian-arm64/buildchroot-target/1.0-r0/rootfs/usr/lib/aarch64-linux-gnu/libz.so.1
```

How can I configure dpkg-shlibdeps to find the needed library?

Thanks in advance,
Stefano.

jan-kiszka, 2023-05-09:
Just asked colleagues who did .NET packaging for a different Isar image before, but it seems there is room for improvements:
```
#!/usr/bin/make -f
%:
	dh $@

override_dh_strip:
	true

override_dh_shlibdeps:
	true
```
That's their custom rules files used with a very similar recipe, but one that does not specify `DEBIAN_DEPENDS` at all. More details expected, will keep you posted.

SCordibella, 2023-05-10:
Thank you @jan-kiszka for the hint. I add a rules file and now I am able to create the image.
Next step is to test with a real application. 
If you have any improvements for this workaround please share.

Are there any documentation about the rules files in ISAR and generally speaking about the debian policy (I am new to debian development)?

jan-kiszka, 2023-07-19:
Closing as we have at least a workaround.

closed 2023-07-19 by jan-kiszka