siemens/meta-iot2050 #441: Can't find the root device with matching UUID!

SCordibella, 2023-05-15:
Dear all, I am trying to run a custom swupdate-image on a real device.
Unforunately my image doesn't boot and I have the following error (full log [here](https://github.com/siemens/meta-iot2050/files/11475610/full-log.txt)):
```
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... [    4.867698] EXT4-fs (mmcblk0p6): mounted filesystem with ordered data mode. Opts: (null)
[    4.916963] EXT4-fs (mmcblk0p7): mounted filesystem with ordered data mode. Opts: (null)
Begin: Waiting for IMAGE_UUID=cd87ddf8-0c1c-46e3-ae41-a0c653316ce9 ... [   13.389476] tidss 4a00000.dss: [drm] Cannot find any crtc or sizes
[   16.783164] random: crng init done

[   33.867190] tps62363-vout: disabling
done.
Can't find the root device with matching UUID!
(initramfs) (initramfs) (initramfs) 
```
My image recipe just inherit from swu-example and add some packages.

It could be related to [this](https://github.com/siemens/meta-iot2050/commit/e1a4587946a7a8ac79623427e7450f649100f9b3)? 

Best regards,
Stefano.

jan-kiszka, 2023-05-15:
Initrd and rootfs need to be in lock-step. Your initrd is carrying the UUID cd87ddf8-0c1c-46e3-ae41-a0c653316ce9, but none the rootfs have that (you find it in /etc/os-release there). This might be due to some transient build issues, e.g. when doing a partial rebuild (or was this a clean build from scratch?). Or it could be caused by the customizations you did on top of meta-iot2050. I assume you tried a original swupdate image from this layer already as well and found it working?

jan-kiszka, 2023-06-05:
Let's close this for now. If you find out more details about the issue without bbappend, feel free to reopen.

closed 2023-06-05 by jan-kiszka