siemens/meta-iot2050 #460: WWAN interface changes name

rFond, 2023-06-28:
Hi,

I have a problem with the wwan interface which changes its name randomly. Is it possible to freeze this (like the serial port X30) ?

Thank's

jan-kiszka, 2023-07-07:
Which interface of which driver? And to which service is the naming change an issue?

There is always the option to have an own udev rule to pin the interface name, but there also other ways to keep track of the same interface, e.g. MAC- based interface selection in NetworkManager.

rFond, 2023-07-07:
I use mpcie interface with a 4G modem (all it's work correctly). When I do "ifconfig" randomly name of interface "wwan0" in (more of time) change for exemple to "xwwxcodm0".

It's a detail, but if I can fix its more.

For X30 for exemple you use udev rules ? (for me MAC Address change in function of IOT)

jan-kiszka, 2023-07-07:
https://github.com/siemens/meta-iot2050/blob/master/recipes-core/customizations-example/files/20-assign-ethernet-names.rules

jan-kiszka, 2023-07-19:
Please close if that helped.

closed 2023-07-20 by rFond