siemens/meta-iot2050 #462: Arduino SPI pinmux error

andyliao5555, 2023-07-04:
I made an OS by myself, which is V01.03.01-213-g7ffd195, when I use the iot2050setup tool to configure the Arduino SPI pinmux, the SPI is not available.
I connected the IO11-IO12 which means loopback test.
On  V01.03.01-213-g7ffd195:

iot2050setup->Peripherals->Configure Arduino I/O->Enable SPI on IO0-IO3

print pincontroller:
root@iot2050-debian:/#
root@iot2050-debian:/# busybox devmem 0x4301C09C
0x00010000
root@iot2050-debian:/# busybox devmem 0x4301c094
0x00050000
root@iot2050-debian:/# busybox devmem 0x4301c098
0x00050000
root@iot2050-debian:/# busybox devmem 0x4301c090
0x00050007
root@iot2050-debian:/#

do the spi loopback test:
root@iot2050-debian:/home# ./spidev_test -D /dev/spidev0.0 -v
spi mode: 0x0
bits per word: 8
max speed: 500000 Hz (500 KHz)
TX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@....▒..................▒.
RX | 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D  | ................................
root@iot2050-debian:/home#

we can see from the result，the spi loopback failed, RX and TX not the same.
I checked the device tree from k3-am65-iot2050-common.dtsi, found the spi pinmux was wrong, 7 was GPIO.
d13_spi0_clk: d13-spi0-clk {
		pinctrl-single,pins = <
			/* (Y1) MCU_SPI0_CLK */
			AM65X_WKUP_IOPAD(0x0090, PIN_INPUT, 7)
		>;
	};

	d13_gpio: d13-gpio {
		pinctrl-single,pins = <
			/* (Y1) WKUP_GPIO0_48 */
			AM65X_WKUP_IOPAD(0x0090, PIN_INPUT, 7)

so I changed the spi clk pincontroller by busybox devmem 0x4301c090 32 0x00050000 and do the test again:
root@iot2050-debian:/home# ./spidev_test -D /dev/spidev0.0 -v
spi mode: 0x0
bits per word: 8
max speed: 500000 Hz (500 KHz)
TX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@....▒..................▒.
RX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@....▒..................▒.
root@iot2050-debian:/home#

it works!

jan-kiszka, 2023-07-04:
Same question here: regression of https://github.com/siemens/meta-iot2050/commit/d8af4f999cd6999a68872d15060bd69d213c4130?

andyliao5555, 2023-07-05:
> Same question here: regression of [d8af4f9](https://github.com/siemens/meta-iot2050/commit/d8af4f999cd6999a68872d15060bd69d213c4130)?

yes, I‘m verifying the new pinmux implementation based on pinmux-select.

jan-kiszka, 2023-07-06:
@benbrenson

benbrenson, 2023-07-06:
> Same question here: regression of [d8af4f9](https://github.com/siemens/meta-iot2050/commit/d8af4f999cd6999a68872d15060bd69d213c4130)?

I would say that this is more device tree specific, hence f9a9cef is the commit in question. 
Thanks for debugging!

commit 43a077e0 (2023-07-19): linux-iot2050: fix pinmux for SPI CLK on d13

Fixes #462.

Fixes: f9a9ceff7f27 ("linux-iot2050: allow pinmux as non root user")
closed 2023-07-19 by BaochengSu