siemens/meta-iot2050 #463: IOT2050 Advanced PG2 board Display enumeration

arulcse2008, 2023-07-04:
Display is not enumerating properly, 
How to reproduce the issue
1) HotPlug Display Cable after the device boot
2) Display Cable connected but display is turned off.

In these scenarios display won't be enumerating properly, even **/dev/fb0** won't be available
`/dev/fb0` is enumerating only when the display is connected but still the display does not show anything even with or without X installed,

This issue is observed on the latest 5.10.184+cip36 kernel as well. on 5.10.175+cip29

```
root@iot2050-debian:~# dmesg |grep -i clk
[    1.112953] clk: failed to reparent clk:67:0 to clk:67:5: -22
[    1.351698] clk: failed to reparent clk:67:0 to clk:67:5: -22
root@iot2050-debian:~# dmesg |grep -i fb0
[   73.104779] tidss 4a00000.dss: [drm] fb0: tidssdrmfb frame buffer device
root@iot2050-debian:~# uname -a
Linux iot2050-debian 5.10.184-cip36-rt14 #1 SMP PREEMPT_RT Thu, 01 Jan 1970 01:00:00 +0000 aarch64 GNU/Linux
root@iot2050-debian:~# dmesg |grep -i tidss
[    1.355090] [drm] Initialized tidss 1.0.0 20180215 for 4a00000.dss on minor 0
[    1.355442] tidss 4a00000.dss: [drm] Cannot find any crtc or sizes
[    1.357022] tidss 4a00000.dss: [drm] Cannot find any crtc or sizes
[   73.104779] tidss 4a00000.dss: [drm] fb0: tidssdrmfb frame buffer device
root@iot2050-debian:~# 

```

arulcse2008, 2023-07-04:
@jan-kiszka Can you please help us to address this issue, This is observed on sokol-omni distro

jan-kiszka, 2023-07-04:
Did you try multiple DP adapters or - even better - a native DP cable to a monitor already?

arulcse2008, 2023-07-04:
Yeah, We have tried with multiple cables and multiple displays as well. DP --> DP and DP --> mini DP cable as well

jan-kiszka, 2023-07-04:
And is this a regression over previous revisions?

arulcse2008, 2023-07-04:
We can see this failure on 5.10.175 as well, I doubt this behavior is observed in 5.10.104 as well. Kindly refer the logs https://pastebin.alm.mentorg.com/m379ec736

jan-kiszka, 2023-07-04:
You are sharing a link from an internal server.

If you can validate a version that used to work, that would be helpful. I will try to reproduce as well, but I don't have hardware in reach ATM.

arulcse2008, 2023-07-04:
ok, I am also bisecting linux kernel now, will share the details shortly, Thanks

arulcse2008, 2023-07-04:
> We can see this failure on 5.10.175 as well, I doubt this behavior is observed in 5.10.104 as well. Kindly refer the logs https://pastebin.alm.mentorg.com/m379ec736

Kindly refer the logs from below link
https://paste.debian.net/1284913/

BaochengSu, 2023-07-05:
I just verified on 5.10.104, the DP works with a HP E243i DP monitor, when the monitor is connected to the IOT after booting, in power-off status, then power on the monitor.

```
root@iot2050-debian:~# uname -a
Linux iot2050-debian 5.10.104-cip3 #1 SMP PREEMPT Thu Feb 16 13:16:47 UTC 2023 aarch64 GNU/Linux
```

Still verifying the latest master.

BaochengSu, 2023-07-05:
Just verified on latest master head, no issue find... DP works as expected.

```
root@iot2050-debian:~# uname -a
Linux iot2050-debian 5.10.184-cip36 #1 SMP PREEMPT Thu, 01 Jan 1970 01:00:00 +0000 aarch64 GNU/Linux

root@iot2050-debian:~# dmesg |grep -i tidss
[    2.907565] [drm] Initialized tidss 1.0.0 20180215 for 4a00000.dss on minor 0
[    2.915209] tidss 4a00000.dss: [drm] Cannot find any crtc or sizes
[    2.922795] tidss 4a00000.dss: [drm] Cannot find any crtc or sizes
[  115.979590] tidss 4a00000.dss: [drm] fb0: tidssdrmfb frame buffer device
```

Which IOT2050 variant are you on? I am using a PG2 ADVANCED.

arulcse2008, 2023-07-06:
> The Example image is downloaded from https://github.com/siemens/meta-iot2050/actions/runs/5369483547.

The reported issue is observed on the above image as well, when we have installed lxde desktop. Without X issue was not observed.

arulcse2008, 2023-07-06:
> > The Example image is downloaded from https://github.com/siemens/meta-iot2050/actions/runs/5369483547.
> 
> The reported issue is observed on the above image as well when we have installed lxde desktop. Without X issue was not observed.

@BaochengSu @jan-kiszka 

The reported issue is fixed with the below changes

```
root@iot2050-debian:/lib/systemd/system# diff lightdm.service lightdm.service.orig
6,9c6,12
< Conflicts=getty@tty1.service getty@tty7.service
< After=getty@tty1.service getty@tty7.service
< BindsTo=dev-fb0.device
< After=dev-fb0.device
---
> # replaces plymouth-quit since lightdm quits plymouth on its own
> Conflicts=plymouth-quit.service
> After=plymouth-quit.service
>
> # lightdm takes responsibility for stopping plymouth, so if it fails
> # for any reason, make sure plymouth still stops
> OnFailure=plymouth-quit.service
root@iot2050-debian:/lib/systemd/system#
root@iot2050-debian:/lib/systemd/system# diff /lib/udev/rules.d/99-systemd.rules /lib/udev/rules.d/99-systemd.rules.orig
82,83d81
< SUBSYSTEM=="graphics", KERNEL=="fb0", SYMLINK="fb0", TAG+="systemd"
<
root@iot2050-debian:/lib/udev/rules.d#

```
CC: @skrishnakar

jan-kiszka, 2023-07-07:
So it is a boot ordering issue? Will you propose a patch that adds a customization recipe for lxde?

arulcse2008, 2023-07-07:
Yeah, Sure will raise a PR shortly

arulcse2008, 2023-07-07:
> Yeah, Sure will raise a PR shortly

PR raised to fix this issue https://github.com/siemens/meta-iot2050/pull/466