siemens/meta-iot2050 #471: Offline Build with Preempt-RT fails

SeyerleinRichard, 2023-07-20:
Offline Build with Preempt-RT fails with following error:

_ERROR: linux-iot2050-rt-5.10.184-cip36-rt14-r0 do_fetch: Network access disabled through BB_NO_NETWORK (or set indirectly due to use of BB_FETCH_PREMIRRORONLY) but access requested with command /usr/bin/env wget -t 2 -T 30 --passive-ftp -P /build/downloads 'https://git.kernel.org/pub/scm/linux/kernel/git/cip/linux-cip.git/snapshot/linux-cip-5.10.184-cip36-rt14.tar.gz' (for url https://git.kernel.org/pub/scm/linux/kernel/git/cip/linux-cip.git/snapshot/linux-cip-5.10.184-cip36-rt14.tar.gz)
ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/linux-iot2050-rt/5.10.184-cip36-rt14-r0/temp/log.do_fetch.2992
ERROR: Task (/build/../repo/recipes-kernel/linux/linux-iot2050-rt_5.10.184-cip36-rt14.bb:do_fetch) failed with exit code '1'_

Content of log file log.do_fetch.2992:

_DEBUG: Executing python function do_fetch
DEBUG: Searching for debian in paths:
    /build/../work/isar/meta/recipes-kernel/linux/files/iot2050
    /build/../repo/recipes-kernel/linux/linux-iot2050-rt-5.10.184-cip36-rt14-r0/iot2050
    /build/../repo/recipes-kernel/linux/linux-iot2050-rt-5.10.184-cip36-rt14/iot2050
    /build/../repo/recipes-kernel/linux/linux-iot2050-rt/iot2050
    /build/../repo/recipes-kernel/linux/files/iot2050
    /build/../repo/recipes-kernel/linux/iot2050
    /build/../work/isar/meta/recipes-kernel/linux/files/arm64
    /build/../repo/recipes-kernel/linux/linux-iot2050-rt-5.10.184-cip36-rt14-r0/arm64
    /build/../repo/recipes-kernel/linux/linux-iot2050-rt-5.10.184-cip36-rt14/arm64
    /build/../repo/recipes-kernel/linux/linux-iot2050-rt/arm64
    /build/../repo/recipes-kernel/linux/files/arm64
    /build/../repo/recipes-kernel/linux/arm64
    /build/../work/isar/meta/recipes-kernel/linux/files/
    /build/../repo/recipes-kernel/linux/linux-iot2050-rt-5.10.184-cip36-rt14-r0/
    /build/../repo/recipes-kernel/linux/linux-iot2050-rt-5.10.184-cip36-rt14/
    /build/../repo/recipes-kernel/linux/linux-iot2050-rt/
    /build/../repo/recipes-kernel/linux/files/
    /build/../repo/recipes-kernel/linux/
DEBUG: Trying PREMIRRORS
DEBUG: Trying Upstream
DEBUG: Fetching https://git.kernel.org/pub/scm/linux/kernel/git/cip/linux-cip.git/snapshot/linux-cip-5.10.184-cip36-rt14.tar.gz using command '/usr/bin/env wget -t 2 -T 30 --passive-ftp -P /build/downloads 'https://git.kernel.org/pub/scm/linux/kernel/git/cip/linux-cip.git/snapshot/linux-cip-5.10.184-cip36-rt14.tar.gz''
ERROR: Network access disabled through BB_NO_NETWORK (or set indirectly due to use of BB_FETCH_PREMIRRORONLY) but access requested with command /usr/bin/env wget -t 2 -T 30 --passive-ftp -P /build/downloads 'https://git.kernel.org/pub/scm/linux/kernel/git/cip/linux-cip.git/snapshot/linux-cip-5.10.184-cip36-rt14.tar.gz' (for url https://git.kernel.org/pub/scm/linux/kernel/git/cip/linux-cip.git/snapshot/linux-cip-5.10.184-cip36-rt14.tar.gz)
DEBUG: Python function do_fetch finished_

Tested with refspef 73e4b161dd4583900ab0038218e5d0b592218d1b from master branch.

jan-kiszka, 2023-07-20:
Works fine here after a first, fetching build, using this offline kas config:
```
header:
  version: 10

local_conf_header:
  offline: |
    ISAR_USE_CACHED_BASE_REPO = "1"
    BB_NO_NETWORK = "1"
```

Didn't you build an RT kernel online first, at least let it fetch?

SeyerleinRichard, 2023-07-21:
Probably not. I thought I had built the RT kernel online, but apparently not.

So please close the issue.

closed 2023-07-21 by jan-kiszka