siemens/meta-iot2050 #478: connection errors on release v01.03.01

maxvaneck, 2023-09-10:
as stated above the latest release fail to build because of connection errors 

ERROR: python-absl-0.15.0-1-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; unset _PYTHON_SYSCONFIGDATA_NAME; export GIT_PROXY_COMMAND="oe-git-proxy"; export NO_PROXY="*"; export PATH="/work/isar/scripts:/work/isar/bitbake/bin:/usr/sbin:/usr/bin:/sbin:/bin"; export HOME="/tmp/tmpqjo3moqk"; git -c core.fsyncobjectfiles=0 ls-remote https://salsa.debian.org/python-team/packages/python-absl.git  failed with exit code 128, output:
fatal: unable to access 'https://salsa.debian.org/python-team/packages/python-absl.git/': Could not resolve host: salsa.debian.org
i have tested this with multiple networks wth tested internet connection as well as using my phone as a hotspot using 5G . it doesn't matter if i'm selecting build with release debian packages or not . Maybe some urls have changed and have to be updated or something? It used to work  until it suddenly didn't anymore 
 any assistance would be welcome,
  thx in advance ,
Max van Eck

jan-kiszka, 2023-09-10:
Hmm, working for me right now, can you retry?

Maybe salsa had an issue while a lot of the Debian developers were hanging out here at DebConf23.

maxvaneck, 2023-09-11:
just retried now with company internet but no dice . still the same errors about   not being able to connect to salsa.debian.org/python-team/packages/python-absl.git
ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/python-absl-host/0.15.0-1-r0/temp/log.do_fetch.113

jan-kiszka, 2023-09-11:
Is it still a dns error, or rather some connection issue? Can you check outside of kas-container if the issue is present there as well? I've tried to wget the URL only so far, both from local India network as well as from a server in Germany, but both worked fine

maxvaneck, 2023-09-11:
seems the issue is somewhere on my end because i just tried again and everything build fine. thx for the help in any case .much appreciated

maxvaneck, 2023-09-11:
it seems the problem lies with enabling google coral support. because if i enable it the build fail and succeeds if i disable it with the same otions checked otherwise

maxvaneck, 2023-09-13:
turns out that wasn't the entire problem. it works fine on my laptop but my desktop fails on this error both using release v01.03.01

EERROR: buildchroot-target-1.0-r0 do_rootfs_install: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_rootfs_install(d)
     0003:
File: '/work/isar/meta/classes/rootfs.bbclass', lineno: 187, function: do_rootfs_install
     0183:                                     shared=True)
     0184:
     0185:        bb.build.exec_func(cmd, d)
     0186:
 *** 0187:        if (d.getVarFlag(cmd, 'isar-apt-lock') or "") == "release-after":
     0188:            bb.utils.unlockfile(lock)
     0189:    progress_reporter.finish()
     0190:}
     0191:addtask rootfs_install before do_rootfs_postprocess after do_unpack
File: '/work/isar/bitbake/lib/bb/build.py', lineno: 251, function: exec_func
     0247:    with bb.utils.fileslocked(lockfiles):
     0248:        if ispython:
     0249:            exec_func_python(func, d, runfile, cwd=adir)
     0250:        else:
 *** 0251:            exec_func_shell(func, d, runfile, cwd=adir)
     0252:
     0253:    try:
     0254:        curcwd = os.getcwd()
     0255:    except:
File: '/work/isar/bitbake/lib/bb/build.py', lineno: 452, function: exec_func_shell
     0448:    with open(fifopath, 'r+b', buffering=0) as fifo:
     0449:        try:
     0450:            bb.debug(2, "Executing shell function %s" % func)
     0451:            with open(os.devnull, 'r+') as stdin, logfile:
 *** 0452:                bb.process.run(cmd, shell=False, stdin=stdin, log=logfile, extrafiles=[(fifo,readfifo)])
     0453:        finally:
     0454:            os.unlink(fifopath)
     0455:
     0456:    bb.debug(2, "Shell function %s finished" % func)
File: '/work/isar/bitbake/lib/bb/process.py', lineno: 182, function: run
     0178:        if not stderr is None:
     0179:            stderr = stderr.decode("utf-8")
     0180:
     0181:    if pipe.returncode != 0:
 *** 0182:        raise ExecutionError(cmd, pipe.returncode, stdout, stderr)
     0183:    return stdout, stderr
Exception: bb.process.ExecutionError: Execution of '/build/tmp/work/iot2050-debian-arm64/buildchroot-target/1.0-r0/temp/run.rootfs_install_pkgs_download.854' failed with exit code 100:
Reading package lists...
Building dependency tree...
E: Unable to locate package libc6:armhf
E: Unable to locate package crossbuild-essential-armhf
E: Unable to locate package make
E: Unable to locate package debhelper
E: Unable to locate package autotools-dev
E: Unable to locate package ccache
E: Unable to locate package docbook-to-man
E: Unable to locate package automake
E: Unable to locate package devscripts
E: Unable to locate package equivs
E: Unable to locate package gcc
E: Unable to locate package build-essential
WARNING: exit code 100 from a shell command.

ERROR: Logfile of failure stored in: /build/tmp/work/iot2050-debian-arm64/buildchroot-target/1.0-r0/temp/log.do_rootfs_install.854
ERROR: Task (/work/isar/meta/recipes-devtools/buildchroot/buildchroot-target.bb:do_rootfs_install) failed with exit code '1'
ERROR: buildchroot-host-1.0-r0 do_rootfs_install: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
[... 1646 lines cut ...]
E: Failed to fetch http://snapshot.debian.org/archive/debian/20220222T214253Z/pool/main/libe/libencode-locale-perl/libencode-locale-perl_1.05-1.1_all.deb  Unable to connect to snapshot.debian.org:80: [IP: 193.62.202.27 80]
E: Failed to fetch http://snapshot.debian.org/archive/debian/20220222T214253Z/pool/main/libt/libtimedate-perl/libtimedate-perl_2.3300-2_all.deb  Unable to connect to snapshot.debian.org:80: [IP: 193.62.202.27 80]
E: Failed to fetch http://snapshot.debian.org/archive/debian/20220222T214253Z/pool/main/libh/libhttp-date-perl/libhttp-date-perl_6.05-1_all.deb  Unable to connect to snapshot.debian.org:80: [IP: 193.62.202.27 80]
E: Failed to fetch http://snapshot.debian.org/archive/debian/20220222T214253Z/pool/main/libf/libfile-listing-perl/libfile-listing-perl_6.14-1_all.deb  Unable to connect to snapshot.debian.org:80: [IP: 193.62.202.27 80]
E: Failed to fetch http://snapshot.debian.org/archive/debian/20220222T214253Z/pool/main/libh/libhtml-tagset-perl/libhtml-tagset-perl_3.20-4_all.deb  Unable to connect to snapshot.debian.org:80: [IP: 193.62.202.27 80]
E: Failed to fetch http://snapshot.debian.org/archive/debian/20220222T214253Z/pool/main/libu/liburi-perl/liburi-perl_5.08-1_all.deb  Unable to connect to snapshot.debian.org:80: [IP: 193.62.202.27 80]
E: Failed to fetch http://snapshot.debian.org/archive/debian/20220222T214253Z/pool/main/libh/libhtml-parser-perl/libhtml-parser-perl_3.75-1%2bb1_amd64.deb  Unable to connect to snapshot.debian.org:80: [IP: 193.62.202.27 80]
E: Failed to fetch http://snapshot.debian.org/archive/debian/20220222T214253Z/pool/main/libh/libhtml-tree-perl/libhtml-tree-perl_5.07-2_all.deb  Unable to connect to snapshot.debian.org:80: [IP: 193.62.202.27 80]
E: Failed to fetch http://snapshot.debian.org/archive/debian/20220222T214253Z/pool/main/libi/libio-html-perl/libio-html-perl_1.004-2_all.deb  Unable to connect to snapshot.debian.org:80: [IP: 193.62.202.27 80]
[... cut ...]
E: Failed to fetch http://snapshot.debian.org/archive/debian/20220222T214253Z/pool/main/w/wdiff/wdiff_1.2.2-2%2bb1_amd64.deb  Unable to connect to snapshot.debian.org:80: [IP: 193.62.202.27 80]
E: Failed to fetch http://snapshot.debian.org/archive/debian/20220222T214253Z/pool/main/d/devscripts/devscripts_2.21.3%2bdeb11u1_amd64.deb  Unable to connect to snapshot.debian.org:80: [IP: 193.62.202.27 80]
E: Failed to fetch http://snapshot.debian.org/archive/debian/20220222T214253Z/pool/main/x/xml-core/xml-core_0.18%2bnm

maxvaneck, 2023-10-18:
Yes , it only happens if you choose the  use debian packages from release however

jan-kiszka, 2023-10-18:
You should be able to build from stable/V01.03 using release packages, but as we switched to a different distro in master, this is expected to fail for that branch.

jan-kiszka, 2023-10-23:
So, can we close this?

maxvaneck, 2023-10-23:
yes, with the  mention that building the latest release works only when not enabling the "use release packages" option

jan-kiszka, 2023-10-23:
To be resolved soon as well - we are about to freeze the package feed, thus update the related lock file.

closed 2023-10-23 by jan-kiszka