siemens/meta-iot2050 #480: old nodejs version

maxvaneck, 2023-09-15:
greetings,

I am trying to use the preinstalled node red on a self built  example image version 1.3.1 . I need to install some extra packages for extra functionality . unfortunately i am getting the error below. which i have come to understand means that my nodejs version is too old. it is however the latest version available in the repository. soo do I have to install a newer version some other way or do I jut compile from scratch as explained here ?
https://github.com/nodejs/node/blob/main/BUILDING.md#building-nodejs-on-supported-platforms

WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'responselike@3.0.0',
npm WARN EBADENGINE   required: { node: '>=14.16' },
npm WARN EBADENGINE   current: { node: 'v12.22.12', npm: '7.5.2' }

thx in advance,
Max

jan-kiszka, 2023-09-15:
The 1.3.1 image was using Debian 11 (bullseye) as basis, thus had nodejs 12 only. You would have to use current master to get bookworm and nodejs 18. Or you need to select an older version of that node which still worked with nodejs 12.

maxvaneck, 2023-09-15:
i'll try the current master then. will report back

maxvaneck, 2023-09-15:
unfortunately i went back 4 years in releases but that also didn't work now i got
warn EBADENGINE Unsupported engine {
669 warn EBADENGINE   package: 'unique-slug@4.0.0',
669 warn EBADENGINE   required: { node: '^14.17.0 || ^16.13.0 || >=18.0.0' },
669 warn EBADENGINE   current: { node: 'v12.22.12', npm: '7.5.2' }

jan-kiszka, 2023-09-15:
Then let's fix your build of master - our CI is building it fine: Did you try a clean build (`kas-container --isar cleansstate` first)?

maxvaneck, 2023-09-15:
this was a clean slate git clone but yes i ran it and no luck . interestingly if i disable  use release debian packages it seems to build just fine. (still building on my poor laptop)

jan-kiszka, 2023-09-15:
Yes, sorry, that switch for the released package set makes no sense: last release was Debian 11, now we are building Debian 12. The feed will get updated soon.

maxvaneck, 2023-09-20:
final verdict  using the latest version of master worked like a charm. building only worked with the release packages option turned off however

closed 2023-10-04 by jan-kiszka